micro-manager/micro-manager #1439: Micro-Magellan tile overlap issue

mdanderson03, 2022-04-29:
@henrypinkard I noticed when we evaluated tile overlaps for Micro-Magellan that it seemed to work by finding the largest dimension and using that overlap for both dimensions. This works well for a square camera sensor, but the Iris 15 I have is a 5:3 aspect ratio and its quite the difference for the overlapped region.

henrypinkard, 2022-05-06:
I think this is almost certainly a place in the source code where it should say `imageWidth, imageHeight` but instead says `imageWidth, imageWidth`. Not sure where it would be though, and its a bit annoying to try to debug this without a physical microscope. Could you try running one of these [tiled acquisitions](https://pycro-manager.readthedocs.io/en/latest/acq_events.html#xytiled-acquisition) in pycromanager and see if the same problem surfaces? Would help me narrow it down

mdanderson03, 2022-05-06:
Thanks for the response Henry! I hope your company is doing well too. I will do that and let you know what happens. I have been really pushing micromagellan lately and I have noticed a few other issues after we found the tile overlap one. Let me know if you want me to make new threads for each of these things. 
1. Tile overlap
2. In micromagellan acquisition mode, if I have more than one event, if in event 1, I select surface x to get my z points from for a 2D surface. Then it forces event 2 to get its focus points from the same surface rather than the surface y I wanted it to get its points from. 
3. Micromagellan doesnt seem to take a 3D surface if I select the 3D mode where I want it to follow a surface and do +- a certain range with a certain z step size. 
4. When I  generate a micromagellan acquisition event via pycromanager, it sometimes will say surface does not exist even though it does. I literally will manually go to the dropdown menu and select exactly the same surface that was already selected and it will work. Its weird. 

Sorry for so many issues! I really love micromagellan. Its such an easy way to navigate samples and set up surfaces.

henrypinkard, 2022-05-06:
Yeah that would be great if you could make a thread for each one. I'm all for working out these little bugs. Its also especially helpful if you can reproduce them on the demo config, and also if you can record a little gif of the bug so I can see exactly what you mean

mdanderson03, 2022-05-06:
Thanks! When I get gifs made of each thing, I will make a thread for them.

henrypinkard, 2022-05-28:
Maybe this is fixed by #1449?