micro-manager/micro-manager #1460: Automated linting of Java code (again)

marktsuchida, 2022-06-13:
We briefly experimented with running [CheckStyle](https://checkstyle.org/) on PRs (#1206), and disabled it because it was too noisy (#1221).

Now that @nicost has fixed code styling for most of mmstudio (#1457), we need a way to maintain correct styling of Java code in mmstudio.

I think we need both a GitHub Actions workflow that runs on PRs, and a way to easily run the exact same check(s) locally. Without the latter, it's too cumbersome to fix things. So **I think we should set up the local method first**, and simply invoke it from Actions.

The previous attempt used GitHub's [Super-Linter](https://github.com/github/super-linter), but I think that is the wrong tool for our project. It gains us nothing unless we have more than a few languages that we want to lint, and it requires Docker to run locally, which I think is too much to ask everyone to install (especially on Windows). It also won't cleanly support multiple checks with different settings (e.g. different settings for API vs internal classes in mmstudio).

We should just run CheckStyle [directly](https://checkstyle.org/cmdline.html), but make it easy by providing a script.

Two discussion points:

- Is this a good time to set up a pre-commit [hook](https://www.git-scm.com/book/en/v2/Customizing-Git-Git-Hooks), preferably using [pre-commit](https://pre-commit.com/)? Once enabled for a local Git repo, these hooks will run every time you run `git commit`, but can be skipped with `git commit --no-verify` in an emergency. Use of pre-commit (the app) simplifies maintenance and use of the Git pre-commit hook, but does require Python to be installed on the system. Users need to switch on the hook in their clone, so this needs to be advertised.
    - Personally I think it is worth doing this, because if rules are enforced at the source, we won't have to waste time fixing violations. It seems better than asking people to manually run a check script, or be confronted with issues after submitting a PR. Also, if we start using clang-format in the future, the pre-commit hook can automatically fix formatting and commit, if the issues are purely about formatting. This is much simpler than asking people to set up their IDE to format correctly.
    - A very valid argument against this is that it will be [annoying](https://www.thoughtworks.com/insights/blog/pre-commit-don-t-git-hooked) if it is slow to commit. On my (fast, M1) computer, it currently takes 6.6 seconds to run CheckStyle on mmstudio. Perhaps we should limit any use of pre-commit hooks to very fast things (maybe clang-format, but we'll have to see). It's a tradeoff, though; if we're not going to run it unless it's a hook, then we should probably make it a hook.
- The latest version of CheckStyle requires Java 11 to run. We could either require that everybody install Java 11+ (in addition to Java 8 for correct building of MM), or use the CheckStyle [backport](https://checkstyle.org/#Backport) for Java 8.
    - I'm inclined toward using mainstream CheckStyle; it's not hard to install a JDK.

marktsuchida, 2022-06-14:
Also, we absolutely need to disable the (mostly, but not exclusively, Javadoc-related) rules that are not currently followed; otherwise making any change to the code is excruciating, because we need to make sure we do not add a new style error in a sea of existing ones. I'm finding myself in this position as I try to rebase a topic branch on the current main. Rules can be (re-)added later after ensuring the code follows them.

nicost, 2022-06-14:
I was hoping that the Javadoc violations would nudge us to add Javadoc for any file we touch.  It is too much work to go through all the code and add Javadoc, but when we modify a file, it may be nice to get a friendly nudge to add Javadoc.  Not sure how that could work.

Can the pre-commit hook be set up to only run CheckStyle on the files in that commit?  

I am fine with mainstream CheckStyle and Java11.

nicost, 2022-06-14:
> Are you generally okay with the idea of adding a pre-commit hook

Yes!  Thanks for your work on this!.

marktsuchida, 2022-06-15:
Looks like pre-commit only runs on changed files by default; the GitHub action can also be made to do so: https://github.com/pre-commit/action/issues/7

marktsuchida, 2022-07-08:
Regarding the Javadoc checks, it looks like they can be conditionally enabled, so that they are warnings by default (to nag you in IDEs) but ignored in pre-commit and CI checks (to keep down noise).

```xml
<module name="JavadocMethod">
  <property name="severity"
    value="${checkstyle.javadoc.severity}"
    default="error"/>
</module>
```

closed 2022-07-14 by marktsuchida