micro-manager/micro-manager #1465: Acquistion Engine improvement roadmap

henrypinkard, 2022-06-17:
This issue describes a road map for making acquisition engine improvements. These include both practical changes that can be made immediately and longer term aims to build a more powerful and general acquisition engine that still need to be more fleshed out.

# Phase 1: Wire in AcqEngJ to make use of its existing features
The immediate goal is to wire up AcqEngJ to various parts of micro-manager where it can enhance functionality. AcqEngJ is more flexible, performant, and much easier to change/improve than the current Clojure acquisition engine. It is already used my Pycro-manager and Micro-Magellan.

Immediate changes include:

- Adding it a back end to the Multi-dimensional Acquisition system in the Micro-Manager GUI, for now as an option in parallel with the current Clojure engine, but with the aim of eventually replacing and retiring that engine.
- Adding it as a backend to LightSheetManager. Probably directly and not through the studio API, since in may not fit into that model so neatly

Doing so will provide many many immediate benefits, owing to shortcomings in the existing  Clojure acquisition engine and saving formats. Many of these limitations pointed out by @marktsuchida in https://github.com/micro-manager/mmCoreAndDevices/issues/180:

- Get rid of special treatment of position/time/Z indices. These can still be used, but aren't baked into the acquisition engine in the same way, making it easier to do more general experiments that vary along different axes.
- No circular dependencies on MMStudio. In fact, no dependencies on MMStudio at all (AcqEngJ sits above MMCoreJ only)
- Access to fast and more flexible (i.e. arbitrary axes) file saving through [NDTiffStorage](https://github.com/micro-manager/NDTiffStorage), which has its own python reader already
- The ability to use "Acquisition hooks" at various stages of the acquisition cycle, which enable injection of customized code that runs synchronously with the acquisition engine. More info on those [here](https://pycro-manager.readthedocs.io/en/latest/acq_hooks.html)
- A full syntax for instructions (acquisition events) that can be spelled out in JSON. This means experiments can be saved/loaded from files, or generated from GUI as it is now
- More transparent hardware sequencing behavior. Events get combined and can be printed to JSON before/during dispatch

Since AcqEngJ is much more flexible than the current engine, much of this will be straightforward and just require translation layers between our current ways of storing acquisition settings and the way AcqEngJ describes experiments (more on this below). One thing that comes to mind that still needs to be sorted out is:

- **Autofocus** is not built in an any way to AcqEngJ right now. Currently, in pycromanager people just do it through acquisition hooks. There will need to be some translation of the MDA settings for autofocus to make this work

#### Additional features that could be easily added
- More fine grained control over sequencing--e.g. run sequencing over z-stack, but not over channels
- Possible performance improvements. @marktsuchida mentioned `lazy sequence generation is inefficient`. Wondering if this is specific to Clojure engine or something more general

# Phase 2: A more general acquisition engine using "Acquisition Protocols"
("Acquisition Protocol" is a working title--open to suggestions for a better name)

After making these immediate practical changes, we can move towards addressing a longstanding problem in various acquisition engines and MMCore: Rigid, fixed assumptions about how the microscope functions.

## Background

Currently, MMCore defines an implicit model of microscope, and this model is built around the structure of a turnkey commercial system for cell biology circa 2005. It makes assumptions like a single XY Stage, a single Z stage, a single camera, etc. This model is hardcoded in the core, through default device roles (e.g. `Core-camera`), as well as in all acquisition engines (more below). 

These "roles" are then called by generic functions in the API. For example, `core.snapImage()` applies to the default camera. The advantage of such an approach is that it allows higher level code to be written generically, without explicit knowledge of the microscope. [This function](https://github.com/micro-manager/AcqEngJ/blob/c2ef88e98b2baf4117d3422fca3a6a37204e1c6d/src/main/java/org/micromanager/acqj/internal/acqengj/Engine.java#L410) in the acquisition engine makes a bunch of calls on devices in these default "roles", greatly simplifying even higher level code for acquiring data (like multi-dimensional acquisitions, or the `Acquistion` class in Pycro-Manager).

MMCore currently defines the following properties for device "roles":

* AutoFocus: Which C++ autofocus device to use. Only has one option
* AutoShutter: True/False for whether to use autoshuttering
* Camera: The default camera device.
* ChannelGroup: The default `config group` 
* Focus: The default Z Stage
* Galvo: A default galvonometer
* SLM: A default Spatial Light Modulator
* Shutter: A default shutter
* XYStage: A default XY stage.

Expanding to more use cases is possible, but hacky or inconvenient. For example, the multi-camera device adapter enables the use of two or more physical cameras as a single logical camera. However, this introduces its own difficulties, and there are limits to how far this can be pushed.

### What do people do now?

By either making their own script, or by using acquisition hooks to modify/work around the rigid assumptions. (@dpshepherd has done a good bit of this and may have insights). There are situations where this is desirable, like controlling some other piece of nonstandard hardware which doesn't and likely won't have a device adapter (like fluidics, micro-manipulators, etc.). But it would be nice to have a a way to configure this and get the best of both worlds: being able to reuse as much code as possible, but still having the flexibility to do customized stuff.

The motivates the idea of **Acquisition Protocols**, which are a new layer of configuration that generalizes this hardcoded concept of how the microscope operates.

## Layers of configuration

Right now there are two layers of configuration:

1. Hardware configuration -- what hardware is present on your microscope, and what roles each of them take
3. Acquisition events -- instructions for how to control devices in those roles

Both of these rely on the fixed assumptions about the other. What is needed is an intermediate layer that generalizes these assumptions.  Then you would have three layers of configuration:

1. Hardware configuration -- what hardware is present on your microscope
2. Acquisition protocol -- how that hardware is controlled. Move this stage, take images on two cameras, etc.
3. Acquisition events -- specific instructions for doing a particular protocol. Move stage to position 123.45 um, take image using exposure 40 ms, etc. 
[... 47 lines cut ...]
__Device__DemoCamera2, .readImage
```

This model could then be fed an event like:

```
{
   'exposure': 100,
   'xy_position': (12.3, 432.1),
   'z_position': 7456.6,
   'preset': 'DAPI'
}

```

Maybe could also add conditionals in here, like:

```
if [camera1]
   __Device__DemoCamera1, .snapImage
else 
   __Device__DemoCamera2, .snapImage
```

@nicost @jondaniels @marktsuchida @bls337 @fadero

marktsuchida, 2022-06-22:
Thanks for the writeup, @henrypinkard!

> * Adding it a back end to the Multi-dimensional Acquisition system in the Micro-Manager GUI, for now as an option in parallel with the current Clojure engine, but with the aim of eventually replacing and retiring that engine.

I 100% support this. I guess the question is what are the necessary API changes (if any)? Are you planning to add a new acquisition API, or keep the old one and make AcqEngJ handle it?

I think it is fine to have a new API (the acquisition API in MMStudio is an area we didn't renovate for 2.0 because we knew the engine had to change eventually, and didn't want to have a short-lived API). But the old API needs to continue to work, at least for a couple of years after the new one becomes full-featured, either by keeping the Clojure engine or by emulating it with AcqEngJ.

And if there is going to be a new API, what is it going to look like? Is AcqEngJ API going to be exposed directly, or will there be new MMStudio APIs that help interface with it?

If you intend for user code to use the AcqEngJ API directly, I might put some work into reviewing it before it becomes official, mostly just to make sure it is evolvable.

> * **Autofocus** is not built in an any way to AcqEngJ right now. Currently, in pycromanager people just do it through acquisition hooks. There will need to be some translation of the MDA settings for autofocus to make this work

Translating to hooks added to AcqEngJ sounds great. Probably cleaner than building into the engine.

> * @marktsuchida mentioned `lazy sequence generation is inefficient`. Wondering if this is specific to Clojure engine or something more general

I wasn't talking about a property of lazy sequences in general. The way lazy sequences are currently used in the Clojure engine is inefficient, because it unnecessarily creates an "acquisition event" object for each frame even if many such frames will simply be run as a single sequence acquisition. For example, if you want a simple 10000-frame time series, the Clojure engine first builds ("lazily") a sequence of 10000 acquisition events. Then it compares each successive event to find that no non-hardware-sequenceable changes occur between them, and merges them. This takes a lot of time if the number of frames is large. The laziness means that (if done correctly) it does not take up memory, but it still takes time to process all those events.

As for Phase 2, I think it's a whole separate discussion. I generally agree about the limitations you are trying to address. It helps to use very concrete examples of things that are currently hard to do and how (concretely) it would be done in the new system. At least that is how I think about these problems, and I think it helps when communicating them. The abstract description is necessary at some point, but it's hard to tell what real problems it is intended to solve (and which ones may be overlooked) without concrete use cases.

nicost, 2022-11-25:
Quick testing with Demo Configuration:

1. Activate new AcqEng under Options.  Go to MDA window. Check "Channels". Click "New".  No new Channel appears.
2. Check "Z-Stacks", set up a Z stack with a few slices and run MDA.  Only two slices appear in the output.
3. Check only "Channels", check 3 channels. Only one image appears in  the output.
4. Check both "Time Points (5, 10 seconds apart), and "Multiple Positions" (4 positions), order Time, Position.  Acquisition happens instantly, i.e. all time points are acquired immediately.  If only "Time Points" is check, the interval is honored.

nicost, 2023-01-04:
Items 1 and 2 are fixed in https://github.com/nicost/micro-manager/tree/acqEngJ.  Item 3 needs clarification about metadata handling in AcqEngJ (are channels handled by index or by name or both?).  Looking into item 4.

nicost, 2023-01-04:
Addendum to item 4: with order Position, Time, the time lapse at the first position happens correctly, all subsequence ones are incorrect (there is no delay between the frames).

I believe items 3 and 4 can only be fixed in AcqEngJ.

nicost, 2023-07-05:
I believe this can be closed with several recent updates.  The new AcqEngJ still needs more extensive testing, but most what I could test is working.

closed 2023-07-05 by nicost

henrypinkard, 2023-07-05:
Hooray!