micro-manager/micro-manager #1477: Running studio in IntelliJ

henrypinkard, 2022-07-19:
After loading the config file in a debug run in IntelliJ, the main GUI fails to load and I get the below stack trace:

```java
2022-07-19T12:00:25.919929 tid15168 [IFO,App] 
                                    [       ] java.lang.NullPointerException in Thread[AWT-EventQueue-0,6,main]
                                    [       ]   at org.micromanager.internal.MainFrame.createCommonActionButtons(MainFrame.java:365)
                                    [       ]   at org.micromanager.internal.MainFrame.createComponents(MainFrame.java:478)
                                    [       ]   at org.micromanager.internal.MainFrame.<init>(MainFrame.java:154)
                                    [       ]   at org.micromanager.internal.MMUIManager.createMainWindow(MMUIManager.java:226)
                                    [       ]   at org.micromanager.internal.MMStudio.<init>(MMStudio.java:427)
                                    [       ]   at MMStudioPlugin$1.run(MMStudioPlugin.java:84)
                                    [       ]   at java.awt.event.InvocationEvent.dispatch$$$capture(InvocationEvent.java:311)
                                    [       ]   at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java)
                                    [       ]   at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:756)
                                    [       ]   at java.awt.EventQueue.access$500(EventQueue.java:97)
                                    [       ]   at java.awt.EventQueue$3.run(EventQueue.java:709)
                                    [       ]   at java.awt.EventQueue$3.run(EventQueue.java:703)
                                    [       ]   at java.security.AccessController.doPrivileged(Native Method)
                                    [       ]   at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80)
                                    [       ]   at java.awt.EventQueue.dispatchEvent(EventQueue.java:726)
                                    [       ]   at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
                                    [       ]   at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
                                    [       ]   at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
                                    [       ]   at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
                                    [       ]   at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
                                    [       ]   at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

```

Any ideas what might be causing this and what to try to resolve it?

marktsuchida, 2022-07-19:
Maybe `-Dforce.annotation.index=true`?
https://micro-manager.org/Using_IntelliJ

henrypinkard, 2022-07-19:
Ah yes, that did it! Thanks @marktsuchida !

closed 2022-07-19 by henrypinkard

henrypinkard, 2022-07-19:
*Along with the `File->Settings->Build,Execution,Deployment->Annotation Processors ` also described in that link