micro-manager/micro-manager #1480: Magellan: Issues copying New Grids to MM PositionLIst

nicost, 2022-07-28:
Issue 1: It is not possible to select multiple rows when copying Grids to the MM position list.  This makes it cumbersome when selecting many sites and wanting them in the MM positionlist.

Issue 2: When copying a Grid to the MM positionlist, only the XY coordinates are copied in, not the coordinates of other axes selected in the Positionlist.  Is there currently a way to get those recorded, and/or can those somehow be added to the MM positionlist (we now need to revisit every site and add the Z axis manually)?

Pinging @henrypinkard

nicost, 2022-07-29:
I started working on this in a [branch](https://github.com/nicost/micro-manager/tree/magellan).  Adding in multiple selections is straight forward, however, adding in positions of other selected axes is not.  I guess that information is currently not collected.  Also, XY positions in a grid can be in between other XY Positions, which could necessitate interpolation of the position of other axed.  Hope you have thoughts or pointers @henrypinkard!

henrypinkard, 2022-07-29:
Yes, I think your solution is correct for issue 1. 

For issue 2: Grids are not meant to be associated with anything other than XY positions. If you want to do XYZ positions, you should use surfaces instead. This will probably be faster if your goal is to just generate a bunch of positions an import them to the PositionList because you don't have to keep clicking "New grid". You might have to make a similar method to the export one used for Grids, but this should be easy.

Also, note that since this was all developed when the 2.0 API was in flux, it is entirely independent of the PositionList with the exception of the export function. I think it is best to keep it this way, because most people using Magellan are staying entirely within it--using Magellan's acquisition system instead of MDAs and Surfaces/Grids instead of PositionList. Adding a dependence on the PositionList could be confusing.

> Also, XY positions in a grid can be in between other XY Positions, which could necessitate interpolation of the position of other axed.

I don't quite understand what this means. What are the "other XY Positions"

henrypinkard, 2022-07-29:
On another note, I briefly considered stripping out explore mode as an independent module, which would be fairly straightforward since its almost entirely built on AcqEngJ/NDViewer/NDTiff. This could be built into the main app instead and made to integrate better with the other studio stuff (and not hidden from view behind a plugin with a confusing name)

nicost, 2022-07-30:
> On another note, I briefly considered stripping out explore mode as an independent module, which would be fairly straightforward since its almost entirely built on AcqEngJ/NDViewer/NDTiff. This [...]

I would like that;)  I will play with surfaces, and see if that could help our work flow.  We have sparse samples, and like to use the explore mode to find cells, mark the positions, and export them to MM to run an MDA.  We need the z positions included, since this is with an oil objective, and we need to be in range for the Perfect Focus to engage.  Currently that involves visiting each side, locking the PFS, and replacing the positions with one that includes the current Z.  That is a cumbersome workflow.

I am currently working on making the file I touched pass checkstyle, and will then create a PR for you to review.

henrypinkard, 2022-07-30:
Yeah I'm not surprised that workflow is a bit clunky since I never had that type of hardware while developing it.

Sounds good, and we can talk about how to make it more general and useful

nicost, 2022-08-19:
Issue 1 is addressed in PR #1484.   Issue 2 remains difficult to address.

closed 2022-08-19 by nicost