micro-manager/micro-manager #1487: MM not creating Metadata file when using Multi-D Acqui.

mrjholt, 2022-08-04:
Hello,
We are using MM Studio version 2.0.0; MMCore version 10.1.1. We have recently discovered that when we acquire images using the Multi-dimensional Acquisition (MDA) engine and set images to be automatically saved within the GUI, uMgr does not generate a Metadata file. uMgr will generate a metadata file if a user clicks and selects the image to be manually saved after acquisition is finished; however, this is not something that automatically occurs after ending an MDA despite having the save images function checked within the GUI. Is there anyway to fix this so that the metadata is created when the MDA finishes so that the MDA automatically generates a metadata file in addition to saving the images?

nicost, 2022-08-04:
Do you have Tools > Options - "Create metadata.txt file with Image Stack Files" checked?

nicost, 2022-08-19:
Closing for lack of information. Please re-open if the answer to the question above was yes.

closed 2022-08-19 by nicost