micro-manager/micro-manager #1488: HCW: Help on unavailable devices links to wrong address

nicost, 2022-08-17:
In the Hardware Configuration Wizard, Step 2 of 6, scrolls to the bottom of the device list, select one of the unavailable devices, and click "Help".   You will be directed to a page like: "https://micro-manager.org/wiki/NikonTi2%20(unavailable)" which does not exist.  Stripping "%20(unavailable)" from the address will point to the correct page.

We can either add redirects to all pages, or make sure to strip the "%20(unavailable)" from the web address.  The latter solution seems preferable.

marktsuchida, 2022-08-18:
Please, let's strip it in MMStudio. Although redirecting on the server side has the small advantage of "fixing" existing versions, I don't think it is worth the unmaintainable mess that it will create (it could be made maintainable by building additional tools, but that's so much more work).

nicost, 2022-08-19:
Fixed by stripping the HCW code (in MMStudio) in PR #1489

closed 2022-08-19 by nicost