micro-manager/micro-manager #1491: Shutter remains open after MDA

kosekijkk, 2022-08-22:
Whenever an MDA finishes, the shutter for the first channel remains open. For example, if I do an MDA of multiple positions in the following order:
1. Brightfield
2. GFP

then after the MDA the brightfield shutter stays on and I have to manually turn it off. Is this a bug or a feature?

I'm running MM version 2.0.1 22020401 on a Windows 10.

Thanks!
[image]

marktsuchida, 2022-08-22:
I think it is most likely a configuration issue.
What are the settings in your channel group? (You can copy & paste from your `.cfg` file.)

kosekijkk, 2022-08-22:
Thank you for your prompt response!

Here is a copy of the relevant channels. I forgot to add that I have a third channel involving NI DAQ boards like below.
1. BF
2. GFP
3. NI DAQ. Camera is using DCAM and shutter is State Device Shutter

# Group: Channel
# Preset: GFP
ConfigGroup,Channel,GFP,HamamatsuHam_DCAM,Binning,2x2
ConfigGroup,Channel,GFP,HamamatsuHam_DCAM,Exposure,300.0090
ConfigGroup,Channel,GFP,Dichroic 1,Label,GFP
ConfigGroup,Channel,GFP,EpiShutter 1,State,1
ConfigGroup,Channel,GFP,DiaShutter,State,0
ConfigGroup,Channel,GFP,TransmittedIllumination 1,Brightness,0
ConfigGroup,Channel,GFP,Core,Shutter,EpiShutter 1
ConfigGroup,Channel,GFP,Shutter-1,State,0
ConfigGroup,Channel,GFP,Shutter-2,State,0
ConfigGroup,Channel,GFP,Transmitted Aperture,DiameterAsPercent,10
ConfigGroup,Channel,GFP,DCam,Binning,2
ConfigGroup,Channel,GFP,DCam,Exposure,10.0000
ConfigGroup,Channel,GFP,DCam,OnCameraCCDXSize,2048
ConfigGroup,Channel,GFP,DCam,OnCameraCCDYSize,2048
ConfigGroup,Channel,GFP,DCam,PixelType,16bit
ConfigGroup,Channel,GFP,DigitalIO,Label,0
ConfigGroup,Channel,GFP,State Device Shutter,State Device,DigitalIO
ConfigGroup,Channel,GFP,Core,Camera,HamamatsuHam_DCAM

# Preset: BF
ConfigGroup,Channel,BF,HamamatsuHam_DCAM,Binning,2x2
ConfigGroup,Channel,BF,HamamatsuHam_DCAM,Exposure,10.0070
ConfigGroup,Channel,BF,Dichroic 1,Label,1-NONE
ConfigGroup,Channel,BF,DiaShutter,State,1
ConfigGroup,Channel,BF,TransmittedIllumination 1,Brightness,100
ConfigGroup,Channel,BF,EpiShutter 1,State,0
ConfigGroup,Channel,BF,Core,Shutter,Shutter-1
ConfigGroup,Channel,BF,Shutter-1,State,0
ConfigGroup,Channel,BF,Shutter-2,State,0
ConfigGroup,Channel,BF,Transmitted Aperture,DiameterAsPercent,41
ConfigGroup,Channel,BF,DCam,Binning,2
ConfigGroup,Channel,BF,DCam,Exposure,10.0000
ConfigGroup,Channel,BF,DCam,OnCameraCCDXSize,2048
ConfigGroup,Channel,BF,DCam,OnCameraCCDYSize,2048
ConfigGroup,Channel,BF,DCam,PixelType,16bit
ConfigGroup,Channel,BF,DigitalIO,Label,0
ConfigGroup,Channel,BF,State Device Shutter,State Device,DigitalIO
ConfigGroup,Channel,BF,Core,Camera,HamamatsuHam_DCAM

# Preset: Raman Scan
ConfigGroup,Channel,Raman Scan,Shutter-1,State,0
ConfigGroup,Channel,Raman Scan,HamamatsuHam_DCAM,Binning,2x2
ConfigGroup,Channel,Raman Scan,HamamatsuHam_DCAM,Exposure,300.0090
ConfigGroup,Channel,Raman Scan,Dichroic 1,Label,1-NONE
ConfigGroup,Channel,Raman Scan,EpiShutter 1,State,0
ConfigGroup,Channel,Raman Scan,DiaShutter,State,0
ConfigGroup,Channel,Raman Scan,TransmittedIllumination 1,Brightness,0
ConfigGroup,Channel,Raman Scan,Transmitted Aperture,DiameterAsPercent,41
ConfigGroup,Channel,Raman Scan,Shutter-2,State,0
ConfigGroup,Channel,Raman Scan,DCam,Binning,2
ConfigGroup,Channel,Raman Scan,DCam,Exposure,480000.0000
ConfigGroup,Channel,Raman Scan,DCam,OnCameraCCDXSize,2048
ConfigGroup,Channel,Raman Scan,DCam,OnCameraCCDYSize,2048
ConfigGroup,Channel,Raman Scan,DCam,PixelType,16bit
ConfigGroup,Channel,Raman Scan,DigitalIO,Label,2
ConfigGroup,Channel,Raman Scan,State Device Shutter,State Device,DigitalIO
ConfigGroup,Channel,Raman Scan,Core,Shutter,State Device Shutter
ConfigGroup,Channel,Raman Scan,Core,Camera,DCam

marktsuchida, 2022-08-22:
Thanks. The first thing to do is to remove shutter `State` properties from your channel group. Instead, you need to ensure Autoshutter is enabled, and let the shutter open/close state be handled automatically. This is done through the `Core`-`Shutter` property, which selects the active shutter for each channel.

So you want `Core`-`Shutter` to have the following values for each channel:
- `GFP`: `Core`-`Shutter` = `EpiShutter 1`
- `BF`: `Core`-`Shutter` = `DiaShutter`
(Not sure what is correct for `Raman Scan`, but hopefully you get the idea.)

The problem with including shutter `State` in the channel group is that now there are two things trying to control the shutter, and the end result is often not what you want.

marktsuchida, 2022-08-22:
You probably also should remove `DigitalIO`-`Label` as this will also conflict with the shutter control.

kosekijkk, 2022-08-22:
Thank you very much! I'll try these suggestions. 

One question though. If I remove `DigitalIO`-`Label`, how do I specify which line to use on the DAQ board? Label 2 means 00010 in my case, so it sends a signal from the second DigitalIO line.

nicost, 2022-08-22:
Is DigitalIO a NIDAQ device?  If so, you should indeed leave it in the channel description.  It is the state device controller by your DigitalIO State device shutter.

In general, it is really helpful to keep your channel descriptions as sparse as possible.  For instance, I would take out camera binning, camera size, pixeltype and exposure time settings.  Micro-Manager itself can be helpful remembering the exposure times per channel (check Tools > Options > "Sync exposure beteern Main and MDA Windows").  YOu can set the others either in System-Startup or in another group with useful camera settings.

marktsuchida, 2022-08-22:
It looks like `DigitalIO` to be the device wrapped by `State Device Shutter`, and the `State Device Shutter` is only the active shutter in one channel.

So the line need not be set in the channel group; instead, I would recommend putting it in the special configuration group named `System`, with preset `Startup`. This gets applied automatically when the configuration is loaded, so it's a good place to put settings that will remain constant the whole time.

If you are using other lines of `DigitalIO` for things other than the (single) shutter, then it gets a little more complicated.

kosekijkk, 2022-08-23:
This issue seems to have been fixed by reducing settings like the below. Thank you so much!

\# Preset: BF
ConfigGroup,Channel,BF,Dichroic 1,Label,1-NONE
ConfigGroup,Channel,BF,Core,Shutter,DiaShutter
ConfigGroup,Channel,BF,Core,Camera,HamamatsuHam_DCAM

\# Preset: Raman Scan
ConfigGroup,Channel,Raman Scan,Dichroic 1,Label,1-NONE
ConfigGroup,Channel,Raman Scan,Core,Shutter,State Device Shutter
ConfigGroup,Channel,Raman Scan,Core,Camera,DCam

\# Group: System
\# Preset: Startup
ConfigGroup,System,Startup,HamamatsuHam_DCAM,Binning,2x2
ConfigGroup,System,Startup,Olympus IX83,FrontPanel,0
ConfigGroup,System,Startup,TransmittedIllumination 1,Brightness,100
ConfigGroup,System,Startup,Transmitted Aperture,DiameterAsPercent,48
ConfigGroup,System,Startup,DCam,OnCameraCCDXSize,2048
ConfigGroup,System,Startup,DCam,OnCameraCCDYSize,2048
ConfigGroup,System,Startup,DCam,PixelType,16bit
ConfigGroup,System,Startup,DigitalIO,Label,2
ConfigGroup,System,Startup,DCam,Binning,2
ConfigGroup,System,Startup,State Device Shutter,State Device,DigitalIO
ConfigGroup,System,Startup,DigitalIO,ClosedPosition,0
ConfigGroup,System,Startup,Shutter-2,State,1
ConfigGroup,System,Startup,Shutter-1,State,1

marktsuchida, 2022-08-23:
Great to hear!

closed 2022-08-23 by marktsuchida