micro-manager/micro-manager #1492: NI DAQ device adapter sends TTL twice during MDA

kosekijkk, 2022-08-22:
When I use the NI DAQ device adapter during MDA, it sometimes sends two TTL signals out when it's supposed to send only one. So far I’m not able to find a reproducible condition, but it seems that once it’s working (TTL sent only once), then it works all the time unless I restart MM or the computer. A typical MDA (multi-position/channel) is the following:
1. Brightfield
2. GFP
3. NI DAQ

Whenever this happens, I'll try logging out/logging in, and most of the time it’ll start working again. It’s been a mystery for some time, so I was curious if anyone else had the same issue.

I’m running MM version 2.0.1 22020401 on a Windows 10 with an NI DAQ PCIe-6353 (https://www.ni.com/en-us/support/model.pcie-6353.html).

Thanks!

marktsuchida, 2022-08-22:
I assume this is related to the setup in #1491. Can you try simplifying your channel group (as I suggested in that issue) and see if this still happens?
If it does, we'll need to see the full setting (property values) of your NIDAQ device as well as the State Device Shutter.

kosekijkk, 2022-08-23:
This still happens even after reducing the settings. Can you take a look at the entire setting? Thanks!

[raman.cfg.txt](https://github.com/micro-manager/micro-manager/files/9410179/raman.cfg.txt)

marktsuchida, 2022-08-23:
Thanks. Looks like this is NI100X, not NIDAQ.

I can take a look at this using a USB-6356. When, during the MDA, does the extra pulse happen?

nicost, 2022-08-23:
Can you please move this to image.sc and post your complete config file?

kosekijkk, 2022-08-23:
Ok so actually my bad, the extra pulse isn't happening anymore! Changing the config file seemed to have solved this. 
I had another problem in my Matlab code which was ultimately causing the same error, but at this point, this is a Matlab problem so I will close this issue. 

Mark and Nico, thank you very much for the support!

closed 2022-08-23 by kosekijkk