micro-manager/micro-manager #1493: Channel needs to be selected twice to capture image right after starting MM

kosekijkk, 2022-08-22:
Right after I start MM, I have to select a channel twice to get the live mode working. The first time I click a channel (e.g. brightfield) and do live mode, I get a blank dark noise image from the camera. If I click again, I get the right image. Is this a normal behavior? I can hear the filters etc. moving after the second click, so it seems like the operation doesn’t fully finish after the first click.
This happens only once after starting MM.

I’m using MM version 2.0.1 22020401 on a Windows 10 with an Orca Flash 4.0 v2 camera.

Thanks!

marktsuchida, 2022-08-22:
What are your channel group settings now? The best way to troubleshoot is to find the smallest set of setting in the channel group to reproduce the issue.

marktsuchida, 2022-08-23:
Yeah, I still suspect the State and Label properties conflicting with autoshutter control.

nicost, 2022-08-23:
Remove everything that is unchanged between these two channel definitions.  Remove all the Shutter-State instances.  B.t.w., this is much more a configuration issue rather than a code related issue, and as such makes more sense on https://image.sc (tag with "micro-manager").

kosekijkk, 2022-08-23:
I tried trimming down my channel definition like below, but now I have to click on "Startup" to get BF working. Was this the proper way to configure things? Also, sorry for posting in the wrong place! I wasn't sure if this was a code issue or not.
 [New Recording - 8/23/2022, 1:53:56 PM](https://vimeo.com/742365411/967513c7ac)

\# Preset: BF
ConfigGroup,Channel,BF,HamamatsuHam_DCAM,Exposure,10.0070
ConfigGroup,Channel,BF,Dichroic 1,Label,1-NONE
ConfigGroup,Channel,BF,Core,Shutter,Shutter-1
ConfigGroup,Channel,BF,DCam,Exposure,10.0000
ConfigGroup,Channel,BF,Core,Camera,HamamatsuHam_DCAM

\# Group: System
\# Preset: Startup
ConfigGroup,System,Startup,HamamatsuHam_DCAM,Binning,2x2
ConfigGroup,System,Startup,Olympus IX83,FrontPanel,0
ConfigGroup,System,Startup,TransmittedIllumination 1,Brightness,100
ConfigGroup,System,Startup,Transmitted Aperture,DiameterAsPercent,48
ConfigGroup,System,Startup,DCam,OnCameraCCDXSize,2048
ConfigGroup,System,Startup,DCam,OnCameraCCDYSize,2048
ConfigGroup,System,Startup,DCam,PixelType,16bit
ConfigGroup,System,Startup,DigitalIO,Label,2
ConfigGroup,System,Startup,DCam,Binning,2
ConfigGroup,System,Startup,State Device Shutter,State Device,DigitalIO
ConfigGroup,System,Startup,DigitalIO,ClosedPosition,0
ConfigGroup,System,Startup,Shutter-2,State,1
ConfigGroup,System,Startup,DiaShutter,State,1

nicost, 2022-08-23:
System - Startup should be applied when MM starts up, so there should be no need to click on it.  I do believe that 

`ConfigGroup,System,Startup,DigitalIO,Label,2`

does belong in your channel configuration.  

It is not advised to add exposure times to the channels.  Set them in the UI, and MM will remember.  Also, the MDA has settings for exposure that will interfere with the setting in your channel definition.

kosekijkk, 2022-08-23:
So with the current configuration, the only way to get BF working was to click on System - Startup which seemed a bit weird. There are some settings that were applied properly during startup (e.g. binning), but not the shutter state until I clicked on it.

I'll move DigitalIO to channel config per your suggestion!

nicost, 2022-08-23:
Ah, you do need something like
`ConfigGroup,Channel,BF,Core,Shutter,EpiShutter`
and make sure that AutoShutter is checked the main window.

marktsuchida, 2022-08-23:
@nicost I'm not seeing why DigitalIO-Label should be in the channel group, although it may not hurt either, other than complicating the troubleshooting. DigitalIO-ClosedPosition definitely should not be in the channel group. My understanding is that DigitalIO is being used by State Device Shutter only, and State Device Shutter is being used only in one channel -- so the Label need not change between channels. It would need to be in the channel group if different channels need to use the State Device Shutter to activate different digital lines.

kosekijkk, 2022-08-23:
Ok, it seems like `Shutter-1` doesn't work the same way as `DiaShutter` does (`Shutter-1` is a Prior shutter, and `DiaShutter` is Olympus's original shutter). Are there other things I need to do for Prior shutters?
I prefer the Prior one because it open/closes faster than the original. I attached my whole cfg file below.
Thanks!

[raman.cfg.txt](https://github.com/micro-manager/micro-manager/files/9410143/raman.cfg.txt)

marktsuchida, 2022-08-23:
I don't see anything wrong with the .cfg.
Does the Prior `Shutter-1` work with the Open/Close button in the main window (when Autoshutter is unchecked)?

nicost, 2022-08-23:
I am assuming that  DigitalIO-Label refers to the NIDAQ Digital out, and that the TTL to it controls a light source (such as a laser).  There also needs to be a State-Device-Shutter in the config.  That is how I go about making these work.  Admittedly, a lot of conjecture, but that is because we have never seen the complete config file.  It would be much easier to post that on image.sc, close this issue and continue there.

kosekijkk, 2022-08-23:
Hi Mark and Nico,

Thank you for the suggestions! I created a new post here along with my configuration file. [https://forum.image.sc/t/autoshutter-not-working-with-prior-shutters/70985?u=kosekijkk]

FYI, Shutter-1 did work with the Open/Close button.

marktsuchida, 2022-08-23:
Thanks! Closing here for now.

closed 2022-08-23 by marktsuchida