micro-manager/micro-manager #1499: Policy/future plans for MultipageTIFF vs NDTiff

marktsuchida, 2022-08-23:
Now that NDTiff has been added as a possible save format, how are we going to answer user questions about the difference, or which one should be selected by default?
Do we want to promote the use of NDTiff for all purposes, or only some (and why)?
What are the differences (if any) in stored metadata?

(User feedback also welcome here!)

henrypinkard, 2022-08-23:
I've got a pretty detailed spec on why NDTiff: https://github.com/micro-manager/NDTiffStorage

Perhaps is would be useful to put a table on the MM website specifying performance, downstream use cases, etc

marktsuchida, 2022-08-23:
That sounds like a great idea!

henrypinkard, 2022-08-23:
> What are the differences (if any) in stored metadata?

Shouldn't be any different because its a function of the Acquisition engine + other higher levels APIs. NDTiff's API makes everything required explicit in function arguments, so you don't have to pass in specific metadata to get it to work. The other formats didn't used to have this property--not sure what the current status is.

> Do we want to promote the use of NDTiff for all purposes, or only some (and why)?

For high speed or large (> 10GB ish) datasets, NDTiff is superior. For downstream compatibility, I think it will be hard to pass OME-TIFF. 

> which one should be selected by default?

If you're not writing your own analysis code, OME-Tiff is probably the most widely useful. If you are writing your own analysis code, you are more likely to use pycro-manager, where NDTiff is also the default. I would lean towards not changing the default for MM MDA, because I forsee a lot of complaints when people's usual pipelines break due to the changing format.

marktsuchida, 2022-08-23:
Excellent description! I'm trying to think where it would be good to indicate these facts so that first-time users are likely to see them. (We get enough questions and confusion around the single-page and multi-page TIFF formats and this has bothered me forever.)

henrypinkard, 2022-08-23:
Tooltip on the MDA window? Or "help" link on there to a description?

marktsuchida, 2022-08-23:
Yeah, something like that would probably be nice. I slightly prefer a visible link or button over an extra-long tooltip.

henrypinkard, 2022-08-24:
https://github.com/micro-manager/micro-manager.github.io/pull/107 in the docs
 
and 

https://github.com/micro-manager/micro-manager/pull/1507 for the link

closed 2022-09-06 by marktsuchida