micro-manager/micro-manager #1501: Checkstyle failures

marktsuchida, 2022-08-23:
Checkstyle is finding style issues with the changes made in #1496.
@henrypinkard Would you mind fixing these?
See [doc/pre-commit.md](https://github.com/micro-manager/micro-manager/blob/main/doc/pre-commit.md) for details.

henrypinkard, 2022-08-23:
There seems to be a problem with the instructions for setting this up. When I go to do the final step

> Additionally, import the Checkstyle rules into code formatting settings: go to Settings > Editor > Code Style > Java; from the Gear (⚙️) menu, choose Import Scheme > Checkstyle Configuration.

I get this error:
[image]

Maybe this is related to problems in the codestyle file? ([stakcoverflow](https://stackoverflow.com/questions/57037571/maven-checkstyle-plugin-caused-by-com-puppycrawl-tools-checkstyle-api-checksty))

Latest IntelliJ on Mac

Thoughts?

marktsuchida, 2022-08-23:
Checkstyle 10.3.1 or 10.3.2 from the command line (or pre-commit hook) is working (including on macOS), so the problem is specific to the IDEA plugin and unrelated to the StackOverflow question you linked.

But not sure at the moment how to solve this....

marktsuchida, 2022-08-23:
I can reproduce it on macOS. Previously I had only tried on Windows.

nicost, 2022-08-23:
I ran into this previously (a few days ago) on my laptop whereas it had worked previously, and failed to fix it.  However, I just installed the latest CheckStyle plugin on my desktop, made sure in CheckStyle settings that version 10.3.2 is selected, and then run it without issues.  I'll check again on my laptop...

marktsuchida, 2022-08-23:
It turns out that a more detailed exception log can be [viewed](https://intellij-support.jetbrains.com/hc/en-us/articles/207241085-Locating-IDE-log-files).

<details><summary>full stack trace</summary>

```
2022-08-23 12:49:10,816 [1037797]   WARN - #o.i.i.c.i.CheckStyleCodeStyleImporter - Failed to import style
org.infernus.idea.checkstyle.exception.CheckstyleServiceException: Error executing command 'OpLoadConfiguration': internal error - Failed to call com.puppycrawl.tools.checkstyle.ConfigurationLoader.loadConfiguration()
	at org.infernus.idea.checkstyle.service.CheckstyleActionsImpl.executeCommand(CheckstyleActionsImpl.java:147)
	at org.infernus.idea.checkstyle.service.CheckstyleActionsImpl.loadConfiguration(CheckstyleActionsImpl.java:111)
	at org.infernus.idea.checkstyle.importer.CheckStyleCodeStyleImporter.loadConfiguration(CheckStyleCodeStyleImporter.java:83)
	at org.infernus.idea.checkstyle.importer.CheckStyleCodeStyleImporter.importScheme(CheckStyleCodeStyleImporter.java:52)
	at org.infernus.idea.checkstyle.importer.CheckStyleCodeStyleImporter.importScheme(CheckStyleCodeStyleImporter.java:22)
	at com.intellij.application.options.codeStyle.CodeStyleSchemesActions.importExternalCodeStyle(CodeStyleSchemesActions.java:116)
	at com.intellij.application.options.codeStyle.CodeStyleSchemesActions.chooseAndImport(CodeStyleSchemesActions.java:87)
	at com.intellij.application.options.codeStyle.CodeStyleSchemesActions.importScheme(CodeStyleSchemesActions.java:50)
	at com.intellij.application.options.schemes.AbstractSchemeActions$ImportAction.actionPerformed(AbstractSchemeActions.java:301)
	at com.intellij.openapi.actionSystem.ex.ActionUtil.lambda$performActionDumbAwareWithCallbacks$4(ActionUtil.java:235)
	at com.intellij.openapi.actionSystem.ex.ActionUtil.performDumbAwareWithCallbacks(ActionUtil.java:256)
	at com.intellij.openapi.actionSystem.ex.ActionUtil.performActionDumbAwareWithCallbacks(ActionUtil.java:235)
	at com.intellij.ui.popup.ActionPopupStep.performAction(ActionPopupStep.java:240)
	at com.intellij.ui.popup.ActionPopupStep.performAction(ActionPopupStep.java:249)
	at com.intellij.ui.popup.ActionPopupStep.performAction(ActionPopupStep.java:245)
	at com.intellij.ui.popup.ActionPopupStep.lambda$onChosen$1(ActionPopupStep.java:220)
	at com.intellij.openapi.application.TransactionGuardImpl.performActivity(TransactionGuardImpl.java:106)
	at com.intellij.openapi.application.TransactionGuardImpl.performUserActivity(TransactionGuardImpl.java:95)
	at com.intellij.ui.popup.AbstractPopup.lambda$dispose$18(AbstractPopup.java:1538)
	at com.intellij.util.ui.EdtInvocationManager.invokeLaterIfNeeded(EdtInvocationManager.java:101)
	at com.intellij.ide.IdeEventQueue.ifFocusEventsInTheQueue(IdeEventQueue.java:179)
	at com.intellij.ide.IdeEventQueue.executeWhenAllFocusEventsLeftTheQueue(IdeEventQueue.java:132)
	at com.intellij.openapi.wm.impl.FocusManagerImpl.doWhenFocusSettlesDown(FocusManagerImpl.java:172)
	at com.intellij.ui.popup.AbstractPopup.dispose(AbstractPopup.java:1535)
	at com.intellij.ui.popup.WizardPopup.dispose(WizardPopup.java:162)
	at com.intellij.ui.popup.list.ListPopupImpl.dispose(ListPopupImpl.java:326)
	at com.intellij.ui.popup.PopupFactoryImpl$ActionGroupPopup.dispose(PopupFactoryImpl.java:272)
	at com.intellij.openapi.util.ObjectTree.runWithTrace(ObjectTree.java:139)
	at com.intellij.openapi.util.ObjectTree.executeAll(ObjectTree.java:169)
	at com.intellij.openapi.util.Disposer.dispose(Disposer.java:219)
	at com.intellij.openapi.util.Disposer.dispose(Disposer.java:207)
	at com.intellij.ui.popup.WizardPopup.disposeAllParents(WizardPopup.java:266)
	at com.intellij.ui.popup.WizardPopup.disposeAllParents(WizardPopup.java:268)
	at com.intellij.ui.popup.list.ListPopupImpl.handleNextStep(ListPopupImpl.java:434)
	at com.intellij.ui.popup.list.ListPopupImpl._handleSelect(ListPopupImpl.java:406)
	at com.intellij.ui.popup.list.ListPopupImpl.handleSelect(ListPopupImpl.java:361)
	at com.intellij.ui.popup.list.ListPopupImpl$MyMouseListener.mouseReleased(ListPopupImpl.java:618)
	at java.desktop/java.awt.AWTEventMulticaster.mouseReleased(AWTEventMulticaster.java:298)
	at java.desktop/java.awt.Component.processMouseEvent(Component.java:6654)
	at java.desktop/javax.swing.JComponent.processMouseEvent(JComponent.java:3345)
	at com.intellij.ui.popup.list.ListPopupImpl$MyList.processMouseEvent(ListPopupImpl.java:694)
	at java.desktop/java.awt.Component.processEvent(Component.java:6419)
	at java.desktop/java.awt.Container.processEvent(Container.java:2263)
	at java.desktop/java.awt.Component.dispatchEventImpl(Component.java:5029)
	at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2321)
	at java.desktop/java.awt.Component.dispatchEvent(Component.java:4861)
	at java.desktop/java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4918)
	at java.desktop/java.awt.LightweightDispatcher.processMouseEvent(Container.java:4547)
	at java.desktop/java.awt.LightweightDispatcher.dispatchEvent(Container.java:4488)
	at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2307)
	at java.desktop/java.awt.Window.dispatchEventImpl(Window.java:2793)
	at java.desktop/java.awt.Component.dispatchEvent(Component.java:4861)
	at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:778)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:727)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:721)
	at java.base/java.security.AccessController.doPrivileged(Native Method)
[... cut ...]
	at com.puppycrawl.tools.checkstyle.ConfigurationLoader.loadConfiguration(ConfigurationLoader.java:288)
	... 130 more
```

</details>

So it looks similar to https://github.com/jshiell/checkstyle-idea/issues/480. The exception probably indicates that the plugin thinks that severity is set to `""` somewhere.
My first thought was to try setting `checkstyle.javadoc.severity` explicitly in the file, but that did not help. I do not see any place where severity is being set to any other property reference, so stuck for now.

nicost, 2022-08-23:
I believe that you set those when installing the plugin.  I surely left those empty (as no options are offered).  There may be a way to set them after the fact, but you can try uninstalling and re-installing the plugin.

marktsuchida, 2022-08-23:
I just updated IDEA and the Checkstyle plugin but it still doesn't work (on macOS). Will try reinstalling the plugin.

marktsuchida, 2022-08-23:
(You can edit the property values by editing the entry for our config file under the tool settings. But those settings (logically) shouldn't apply when importing the coding style (which is a more-or-less independent feature). That's why I tried setting in the XML file itself, but it didn't help.)

nicost, 2022-08-23:
Desktop: Windows 10, IntelliJ Community 2021.3, CheckStyle-IDEA 5.70.0, CheckStyle version 10.3.2 works (without setting key-value pairs).
Laptop: Windows 10, IntelliJ Community 2021.3, CheckStyle-IDEA 5.67.2, CheckStyle verison 10.3, now also works...  Not quite sure what changed, I surely ran into this same issue before.

marktsuchida, 2022-08-23:
macOS IntelliJ IDEA 2022.2.1 (Community Edition), CheckStlye-IDEA 5.70.0, Checkstyle 10.3.2.

marktsuchida, 2022-08-23:
In the meantime, @henrypinkard, if you reformat the code to use 3-space indentation the bulk of Checkstyle errors should go away, and what remains may be few enough to fix manually for the time being.

henrypinkard, 2022-08-23:
https://github.com/micro-manager/micro-manager/pull/1506

closed 2022-08-23 by marktsuchida