micro-manager/micro-manager #1508: Replace Magellan Dependency with licensed version

henrypinkard, 2022-08-26:
@marktsuchida It looks like there is an OSI-licensed of the Magellan dependency we discussed the other day:

https://github.com/yonatang/JDT/blob/master/pom.xml

I'm not sure if the Jar is in any repository, but I could copy the relevant source code into Magellan

marktsuchida, 2022-08-26:
That's excellent. No need to copy code; if it is not available in a Maven repo, we can use [JitPack](https://jitpack.io/) when the time comes (might already be possible).

That leaves DT1.2.jar, I think.

henrypinkard, 2022-08-26:
No this is DT1.2.jar, it just changed the name. The API looks essentially the same though, I'd just need to change a couple function names

marktsuchida, 2022-09-03:
Ah, I see! Excellent.