micro-manager/micro-manager #1509: Fix or disable code formatting check

marktsuchida, 2022-09-06:
See discussion in #1501 regarding lack of way to format with IntelliJ on macOS. (@henrypinkard raised the issue again in #1507.)

What would solve all our woes is having an autoformatter _that can be invoked from the pre-commit hook_, so that the code is always formatted correctly whether the IDE is configured or not (I tried to bring this up when we first started looking at Checkstyle, but probably failed to emphasize how important it could be). Unfortunately, I looked into this earlier this year and did not find an immediately attractive solution:
- google-java-format is not suitable for our project because it only allow their specific formatting style.
- Eclipse's formatter is well regarded and some other IDEs (including IntelliJ) have plugins for it. But it cannot be run from the command line (note 1).
- clang-format supports Java, but didn't seem to be as "good" as the IDE formatters (such as IntelliJ). Also it is not as widely supported by IDEs.

note 1: There is a Maven plugin for Eclipse's formatter. Maybe we could hack something that generates a temporary pom.xml just for the formatting, but I don't know how easy that is. We need to limit it to the changed files (otherwise long waits on every commit), which doesn't seem easy.

In any case, we'd need to come up with a style definition for whatever formatter we choose that is compatible with our Checkstyle style -- or remove the formatting-related rules from the latter.

So with that option not looking easy, what shall we do for now?
- We could turn off the check, but then somebody has to fix things later somehow (by re-saving the files in a working IntelliJ or by waiting until we have an autoformatter).
- For IntelliJ on macOS, we could manually configure its coding style until it is close enough, so that if any manual fixes are needed they are few enough to be easy.
- Can IntelliJ export its own style rules? If so, we could export the style from Windows IntelliJ (after loading the Checkstyle style) and import it on macOS, hopefully without the kind of error seen in #1501. The exported style could be saved in `buildscripts/codestyle` and we make it a rule to always (manually) update it when the Checkstyle definitions change (which should be rare).

marktsuchida, 2022-11-05:
Just discovered a tool that can fix (at least some) Checkstyle violations automatically:
https://docs.openrewrite.org/tutorials/running-recipes/automatically-fix-checkstyle-violations
Unfortunately (as is all too common with Java tools) it cannot be run from the command line, and only (practically) works with Maven or Gradle. So something to set up once our Java projects are mavenized.

marktsuchida, 2022-11-05:
I played with OpenRewrite (above link) a little (full setup requires having a `pom.xml` for everything). It works well for enforcing non-formatting style rules (for example, not writing `= null` redundantly when defining a variable). It doesn't seem to work that well for pure formatting rules (such as line length). Also it should be noted that the rewrites it performs may not always be semantically equivalent (for example, it can rewrite `str.equals("blah")` to `"blah".equals(str)`). I suspect most cases are safe, but we ought to examine the rules with some care if running on existing code (in the example, surrounding code may depend on an NPE being thrown when `str` is `null` -- at least in theory).

It may be possible to handle some aspects of formatting with OpenRewrite by using its formatting recipes (which do _not_ get their settings from a Checkstyle config), but I don't know if it is flexible enough to handle all of our Checkstyle rules.

Once we have Maven anyway, it seems better to handle formatting with the Eclipse formatter and remove the formatting rules from `checkstyle.xml` (leaving non-formatting style rules, which are more worthy of human attention).

henrypinkard, 2022-11-18:
I tried it and it applied autoformatting in intelliJ. It still does not seem to help my PR in https://github.com/micro-manager/micro-manager/pull/1547 pass linting checks.

Can we please _please_ disable this check and just make this optional for now? I am burning hours of time trying to get this to work rather than addressing bugs that are actually causing problems for users. I find the whole system really confusing, and if I were a first time contributor I am certain that I would have gotten intimidated or frustrated by the complexity of the system and given up by now. I cannot imagine how making a barrier to entry like this can possibly be worth the benefits of perfectly consistent code formatting

marktsuchida, 2022-11-18:
Can you at least explain what went wrong?

As I said before, I'm not ruling out disabling it, but would rather not do so without knowing what the issue is and reaching agreement that there are in fact unsurmountable issues that cannot be fixed in the short term. So far I have not seen any details from you that indicate that this is the case. I'm not doubting that it is confusing to you, but you need to explain what part remains confusing after our attempts to improve.

I don't agree with the argument that attempts to improve our code style and consistency are less important than addressing bugs. In my view, our messy code formatting causes me headaches every single day, including when I have to review PRs that contain spurious formatting changes or inconsistent style. It's just a different way to improve the project. Some of the checks that Checkstyle performs are actually designed to prevent bugs, and uniform styling also tends to prevent misreading of code. These are ways to try to reduce the number of bugs before we even have to address them in response to "problems for users".

You say you spent hours on this. Do you mean you spent hours just to test #1548? If so, I need help understanding why. If you are including previous attempts, then I don't see how that translates to future newcomers, because some of those problems have been solved (I hope), thanks to your feedback.

As far as I can tell, #1547 currently has mostly indentation and import order errors. I would have expected #1548 to have fixed this. So either importing the style file (#1548) failed with an error, or appeared to succeed but did not actually set the formatting rules (you should see 3-space indents, for example), or you didn't actually reformat the changed code after importing the style, or something else happened. What happened?

Also, are you ever reading the output from Checkstyle?

henrypinkard, 2022-11-18:
I don't have reason that anything will be technically insurmountable. Just that it (from my perspective) adds a lot of complexity, and my concern is that the natural [tendency to bias towards additive changes](https://www.nature.com/articles/s41586-021-03380-y) only serves to make this more complex. There is linting, Checkstyle, pre-commits, IDE-plugins, OS-dependent issues. All of these things I don't totally understand, and while I could put the time in to try to understand them better, I find this process tedious and frustrating, and its not clear that its a good investment of time given the limited scope of changes that I will make to the mmstudio codebase. It's hard for me to say what went wrong, because I'm super confused about how all these things are supposed to fit together. 

> In my view, our messy code formatting causes me headaches every single day, including when I have to review PRs that contain spurious formatting changes or inconsistent style. It's just a different [...]

That is totally fair. I'm not trying to create more work/problems for you, and it is entirely reasonable that something like this that is irrelevant to my workflow is essential to yours.

I don't think there will be a satisfying consensus solution to this issue in the short term.

nicost, 2022-11-19:
Setting up IntelliJ always takes me time, which is annoying, as I don't need any time editing in VIM and running make (or ant) to build).  However, in the longer run it saves me time, because there are many things I can do much more quickly in this IDE.  This is a bit similar. 
I also wonder how difficult it is to follow the instructions in doc/pre-commit.md to import the java-code-style and press Ctrl-Alt-L in the editor (which is more or less all I needed to do). Does not appear to be hard (and if it failed for some reason, lets work on getting the documentation better).

marktsuchida, 2022-12-06:
Superseded by #1583.