micro-manager/micro-manager #1510: Can't run MM on Mac OS 12

henrypinkard, 2022-09-09:
Using recent nightly builds and Mac OS 12.4, I'm unable to run Micro-Manager at all. I get the error: 

`Can't load library .... libMMCoreJ_wrap.jnilib`

I can't seem to figure out what causes this. I tried `chmod`ing the permissions on that file but no difference. It doesn't seem to show up in the Mac OS security settings where I can allow it to run like ImageJ.

Strangely, I can run it when I launch it via pycro-manager headless mode, which suggests to me that it is something about the imageJ launcher, rather than the file itself

marktsuchida, 2022-09-09:
#1036. I think it works if you build everything from scratch. Need signed installer (#396) for GUI app to work smoothly.

marktsuchida, 2024-08-01:
See #2001.

closed 2024-08-01 by marktsuchida