micro-manager/micro-manager #1524: A more general acquisition engine abstraction using "Acquisition Protocols"

henrypinkard, 2022-09-21:
(copied and split out of https://github.com/micro-manager/micro-manager/issues/1465)

("Acquisition Protocol" is a working title--open to suggestions for a better name)

This issue outlines addressing a longstanding problem in various acquisition engines and MMCore: Rigid, fixed assumptions about how the microscope functions.

## Background

Currently, MMCore defines an implicit model of microscope, and this model is built around the structure of a turnkey commercial system for cell biology circa 2005. It makes assumptions like a single XY Stage, a single Z stage, a single camera, etc. This model is hardcoded in the core, through default device roles (e.g. `Core-camera`), as well as in all acquisition engines (more below). 

These "roles" are then called by generic functions in the API. For example, `core.snapImage()` applies to the default camera. The advantage of such an approach is that it allows higher level code to be written generically, without explicit knowledge of the microscope. [This function](https://github.com/micro-manager/AcqEngJ/blob/c2ef88e98b2baf4117d3422fca3a6a37204e1c6d/src/main/java/org/micromanager/acqj/internal/acqengj/Engine.java#L410) in the acquisition engine makes a bunch of calls on devices in these default "roles", greatly simplifying even higher level code for acquiring data (like multi-dimensional acquisitions, or the `Acquistion` class in Pycro-Manager).

MMCore currently defines the following properties for device "roles":

* AutoFocus: Which C++ autofocus device to use. Only has one option
* AutoShutter: True/False for whether to use autoshuttering
* Camera: The default camera device.
* ChannelGroup: The default `config group` 
* Focus: The default Z Stage
* Galvo: A default galvonometer
* SLM: A default Spatial Light Modulator
* Shutter: A default shutter
* XYStage: A default XY stage.

Expanding to more use cases is possible, but hacky or inconvenient. For example, the multi-camera device adapter enables the use of two or more physical cameras as a single logical camera. However, this introduces its own difficulties, and there are limits to how far this can be pushed.

### What do people do now?

By either making their own script, or by using acquisition hooks to modify/work around the rigid assumptions. There are situations where this is desirable, like controlling some other piece of nonstandard hardware which doesn't and likely won't have a device adapter (like fluidics, micro-manipulators, etc.). But it would be nice to have a a way to configure this and get the best of both worlds: being able to reuse as much code as possible, but still having the flexibility to do customized stuff.

The motivates the idea of **Acquisition Protocols**, which are a new layer of configuration that generalizes this hardcoded concept of how the microscope operates.

## Layers of configuration

Right now there are two layers of configuration:

1. Hardware configuration -- what hardware is present on your microscope, and what roles each of them take
3. Acquisition events -- instructions for how to control devices in those roles

Both of these rely on the fixed assumptions about the other. What is needed is an intermediate layer that generalizes these assumptions.  Then you would have three layers of configuration:

1. Hardware configuration -- what hardware is present on your microscope
2. Acquisition protocol -- how that hardware is controlled. Move this stage, take images on two cameras, etc.
3. Acquisition events -- specific instructions for doing a particular protocol. Move stage to position 123.45 um, take image using exposure 40 ms, etc. 

## Advantages of adding this

- It separates out logic of how microscope is controlled from specific instructions for controlling it, which should make things easier to understand/debug
- It could, in many cases, remove the need for users doing high level stuff from even having to interact with the core at all
- Easier to share new designs of microscopes. User doesn't have to get lost in some long script for controlling microscope. they can just copy the acquisition protocol as is, and then use their own events. We can have examples of different protocols: here's one for a diSPIM, here's one for a single objective light sheet etc.
- Separates reusable code relating to the microscope from custom code related to the experiment, rather than having them both in an acquisition hook. For example. settings for changing microcfluidics shouldn't be in the same place as adding another camera port to the microscope
- Ideally for users/applications doing high level stuff, this would allow everything would be accomplished through the acquisition engine--Live/snapping/moving the microscope etc. I think this could greatly simplify the user experience in many ways.

## How to accomplish this

The correct way of doing this is to make changes to the core in tandem that remove default device roles, though I think the changes described here could be temporarily piloted and refined without changes to the core with a few hacks under the hood.

### Backwards compatibility

The default protocol will just be what is currently hardcoded in the core/engine. So if you don't need the features, you never have to mess with them.

### Some very rough, disorganized thoughts on what an Acquisition Protocol might look like

Essentially this would be a brief script, calling device properties and core functions given a set of arguments supplied at runtime

* Call `MicroscopeModel.execute(event)`
  * `event` is a map of key, value pairs corresponding to a particular set of actions on a microscope
   * Maybe organized in such a way that conveys hardware sequencing? 
   * These actions consist of Config groups (i.e. groups of properties), individual properties, and functions 

Generalize the idea of (Device, Property, Value) thruples to include functions in the essential API
  * E.g. `(“DemoCamera”, “.setExposure”, 300)`
* Stored in config file as: 
  * `DemoCamera, .setExposure, [exposure]`
* Here `[exposure]` is a placeholder for a variable with the name exposure to be provided in the event
* Can also explicitly hard code values in config file, to be reused every time
  * `DemoShutter, .setShutterOpen, True`
* And maybe also default arguments that can be overridden
  * `DemoCamera, .setExposure, [exposure]=100`
* This system would be extremely flexible, but also allow acquisitions to be expressed in a very concise way at runtime. Would make for concise programming through the API, and allow GUI control of more complicated stuff via the acquisition engine.

Example model which corresponds to the roles currently hard coded in core/AcqEng

```
__Device__DemoCamera, .setExposure, [exposure]
__Device__DemoXYStage, .setPosition, [xy_position] 
__Device__DemoZStage, .setPosition, [z_position]
__Keyword__ConfigGroup, Channel, [preset]  
__Device__DemoShutter, setOpen, True
__Device__DemoCamera, .snapImage
__Device__DemoShutter, setOpen, False
__Device__DemoCamera2, .readImage
```

This model could then be fed an event like:

```
{
   'exposure': 100,
   'xy_position': (12.3, 432.1),
   'z_position': 7456.6,
   'preset': 'DAPI'
}

```

Maybe could also add conditionals in here, like:

```
if [camera1]
   __Device__DemoCamera1, .snapImage
else 
   __Device__DemoCamera2, .snapImage
```

marktsuchida, 2022-10-14:
I think it makes sense to have an intermediate layer of configuration that maps physical devices, properties, and actions to logical devices, properties, and actions.

I also agree that this can (and should) be prototyped outside of the Core. I think the only functional support needed from the Core is auto-shutter and everything else can be implemented outside of the Core using the basic methods. In fact, in the interest of separation of concerns, I think it makes sense for it to be a permanently separate library outside of the Core -- with some supporting API perhaps added to the Core -- but the exact form might need to be determined after the API details are fleshed out.

I'm not so sure I understand your approach from the given examples.

```text
__Device__DemoCamera, .setExposure, [exposure]
__Device__DemoXYStage, .setPosition, [xy_position] 
__Device__DemoZStage, .setPosition, [z_position]
__Keyword__ConfigGroup, Channel, [preset]  
__Device__DemoShutter, setOpen, True
__Device__DemoCamera, .snapImage
__Device__DemoShutter, setOpen, False
__Device__DemoCamera2, .readImage
```

I guess I'm not seeing how this abstracts the hardware particulars -- it seems to be strongly coupling the physical device choices to the sequence of actions. Might it make sense to have a declarative mapping (e.g. `exposure` -> `DemoCamera, .setExposure`; `xy_position` -> `DemoXYStage, .setPosition`), and then have the acquisition commands just mention `exposure` and `xy_position` so that they is agnostic of physical devices?

A few other thoughts:
- In the above sequence, there needs to be a wait-for-completion before the `setOpen, True`, and before `snapImage`, but not in other places. How will this knowledge be encoded?
- It doesn't matter for early pseudo-code, but I would encourage using YAML, TOML, or JSON format instead of ad-hoc comma-separated. (For that matter, we should probably stop using the comma-separated format for hardware configs -- it is surprisingly hard to parse and write correctly, and having an easily parsed format would allow building better tools to edit/migrate/merge configs.)

henrypinkard, 2022-10-14:
Agreed on your first several points.

> I guess I'm not seeing how this abstracts the hardware particulars -- it seems to be strongly coupling the physical device choices to the sequence of actions. Might it make sense to have a [...]

Agreed. This is kind of what I had in mind (though my example expresses this poorly). The way I think about this, is that a user could essentially write a function for carrying out one "unit" of an acquisition (e.g. a single image, a single timepoint with z-stack, etc). Then the arguments to that function (e.g., the exposure, the z positions, etc) would be specified in separate instructions. There might even be a way to make creation of the files semi-interactive, by having the user actually write out a script containing this function, and then we could have an automated tool that translates this script into the necessary configuration file.

> * In the above sequence, there needs to be a wait-for-completion before the `setOpen, True`, and before `snapImage`, but not in other places. How will this knowledge be encoded?

Not sure yet...this probably relates to other changes we're making in the core and is good to keep in mind

> * It doesn't matter for early pseudo-code, but I would encourage using YAML, TOML, or JSON format instead of ad-hoc comma-separated. (For that matter, we should probably stop using the [...]

Agree and agree. My preference is for YAML