micro-manager/micro-manager #1545: Plugin that "stitches" a Datastore based on the XY location

nicost, 2022-11-05:
I envision this to result in a gigantic image (in terms of XY pixel size), placing all the images in a XYP (could also include C or X) acquisition into their correct location.  Will me memory intensive, but not difficult to do, and quite useful. Datastore could be saved as an NDTiff to allow browsing in MicroMagellan.

henrypinkard, 2022-11-06:
Yeah probably wouldn't be too difficult, although you wouldn't be able to do this through the current Datastore API. Multi-resolution NDTiff uses a [slightly expanded API](https://github.com/micro-manager/NDTiffStorage/blob/main/java/src/main/java/org/micromanager/ndtiffstorage/MultiresNDTiffAPI.java#L42) compared to the [regular NDTiff one](https://github.com/micro-manager/NDTiffStorage/blob/main/java/src/main/java/org/micromanager/ndtiffstorage/NDTiffAPI.java).

Unlike the current Datastore adapter for NDTiff, you'd need to call [the constructor](https://github.com/micro-manager/NDTiffStorage/blob/ffc1aece5a3fcbb8ba6deec5473d6b06fd69daa3/java/src/main/java/org/micromanager/ndtiffstorage/NDTiffStorage.java#L182) with different args making clear that it is multi-res mode. You need `overlapX` `overlapY`, and  `tiled=true`.

The multires NDTiff works with everything laid out in tiles on a regular grid. So if the stage coordinates aren't already laid out as such, you'll need to assume such a grid and crop images accordingly.  Then you call [`putImageMultiRes`](https://github.com/micro-manager/NDTiffStorage/blob/ffc1aece5a3fcbb8ba6deec5473d6b06fd69daa3/java/src/main/java/org/micromanager/ndtiffstorage/MultiresNDTiffAPI.java#L42) to add each tile to the NDTiff dataset:

```java
public Future putImageMultiRes(Object pixels, JSONObject metadata, HashMap<String, Integer> axes,
                                  boolean rgb, int imageHeight, int imageWidth)
```

 The `axes` field must contain `row` and `column` keys mapping to the row and column indices of the grid.

nicost, 2022-11-10:
Note, the Fiji Grid stitching plugin works well with Micro-Manager data sets, see: https://wiki.library.ucsf.edu/display/CMW/Stitching+Images+Acquired+in+Micro-Manager.  That plugin is GPL, so can not be included in MM, but the current workflow maybe good enough.

nicost, 2022-11-10:
B.t.w., would it be useful to add the ability to put images of different resolution into a Datastore to the Datastore api?  Or does that raise too many issues?  Pinging @marktsuchida.

marktsuchida, 2022-11-10:
@nicost Allowing arbitrary images of different sizes/resolution in a `Datastore` is impossible (that would have to be some other interface). Creating a `MultiresDatastore` as a subinterface of `Datastore` might be doable (but messy, because currently `Datasotore` is not in the business of understanding spatial relations between the images it contains; adding that violates separation of concerns (and `Datastore` already does way too many things)).

Alternatively, I could imagine multiple `DataProvider` instances (one for each resolution) being backed by the same multires NDTiff.