micro-manager/micro-manager #1550: Launch MM with specific ZMQ port number

ieivanov, 2022-11-08:
Following up on #1265

@henrypinkard it would be helpful for us to launch a seconds MM version with a specific ZMQ port number. I can currently do that using headless mode in pycromanager, but then we don't get the niceties of the MM GUI. 

Could the ZMQ port be specifies in the MM config file or `MMStartup.bsh`?

henrypinkard, 2022-11-08:
Might be possible. I don't think config file would make sense because it doesn't have to do with hardware, but I could imagine it would be possible to control programmatically if you added something to the studio API and modified this function somehow:
https://github.com/henrypinkard/micro-manager/blob/98e34243a84381eb161418a1abb9dc0d16cce24a/mmstudio/src/main/java/org/micromanager/internal/MMStudio.java#L571

marktsuchida, 2022-11-09:
I think an environment variable (such as `MMSTUDIO_ZMQ_PORT`) would be easiest (easy to control from the program launching MMSutdio) and also more future-proof. I think environment variables can also be set in Windows shortcuts.

Alternatively, or additionally, a Java system property (such as `mmstudio.zmq.port`) could be used; this can be specified in the `ImageJ.cfg` file for a particular copy of MM (on Windows only) using the syntax `-Dmmstudio.zmq.port=1234`.

The problem with using a startup script (or config file) is that it requires the startup script to run before the zmq server is started. This _happens_ to be the case right now, but I can imagine there might come a time when we want to change it, for example to give pycro-manager more control over how MMStudio starts up and loads the hardware config.

ieivanov, 2022-11-09:
Thank you both, that all makes sense.

I don't think I mentioned this, the motivation for this is to allow us to run two independent MM instances for two simultaneous acquisitions on the same microscope. So I don't so much need a specific ZMQ port number, but rather different known port numbers between the two MM instances.

@henrypinkard what happens currently when a second MM instance is launched? The core log says that the second MM still starts a ZMQ server on port 4827 - is that right? Maybe an easy thing to do would be to check if the port is available, and if not, start the server on a port 10 or 100 numbers away. What do you think?

henrypinkard, 2022-11-09:
> @henrypinkard what happens currently when a second MM instance is launched? The core log says that the second MM still starts a ZMQ server on port 4827 - is that right? 

I'd be surprised if there's not a ZMQ error if it launches two on the same port, but I would imagine youd get interleaving and crazy errors if it is able to.

> Maybe an easy thing to do would be to check if the port is available, and if not, start the server on a port 10 or 100 numbers away. What do you think?

Could be possible, but how would you know which port to use? you'd have to look in the core log? To me that seems a bit brittle

Maybe if you're able to say more about why you want to do this, there might be an easier way to set it up

marktsuchida, 2022-11-09:
Stored settings ("user profiles") in Micro-Manager were not designed for two instances of MM running at the same time, so a solution based on them is probably fragile at best -- another reason to suggest an env var and/or system property that can override the stored setting (but the Options dialog should probably show that an override is in use).

I'm not too keen on the server silently selecting a port that the user has not intentionally configured. It's the kind of thing that makes troubleshooting and support much harder.

ieivanov, 2022-11-17:
Thanks again, it sounds like Mark's idea of env var and/or system property would be most robust.

>I'd be surprised if there's not a ZMQ error if it launches two on the same port

Just launching two MM instances, with the Start ZMQ server option checked off doesn't throw errors that I can see - the core log looks regular. I haven't pushed it further.

For this application I need to run two simultaneous acquisitions (with different hardware, channel and z stack settings, etc.) on the same microscope. Our first order solution has been to start up two MM instances, each responsible for one acquisition. The two acquisitions are orchestrated using pycromanager and a NI DAC card which provides exposure triggers to the two cameras.

All of this is so far possible using the headless mode of pycromanager, which allows me to launch a second MM instance with a specific ZMQ port. I can then set up and acquire data from both cameras with no problem. The feature that I miss most in the headless MM instance is the Live Preview. In parallel, I'm also working to get something similar using the Acquisition class and napari as viewer. [This example](https://forum.image.sc/t/pycro-manager-now-works-with-napari/63982) works well, but only for a specific amount of time. I'd like to be able to start and stop the Live Preview at will. I'm adapting [this example](https://pycro-manager.readthedocs.io/en/latest/application_notebooks/PSF_viewer.html) to see if I can get something like that.

henrypinkard, 2022-11-18:
> For this application I need to run two simultaneous acquisitions (with different hardware, channel and z stack settings, etc.) on the same microscope. Our first order solution has been to start up [...]

I'm curious: why did you need to do this in separate instances? Did you run into a limitation, or does this just seem more straightforward? In theory you can interleave pycro-manager `Acquisitions` or use extra axes to combine them all in one

> The feature that I miss most in the headless MM instance is the Live Preview.

I've thought before about making an ability to acquire live into  a pycro-manager acquisition, which I don't think should be too hard. Would this resolve the problem or do you really need to bring the data into python?

> In parallel, I'm also working to get something similar using the Acquisition class and napari as viewer. [This example](https://forum.image.sc/t/pycro-manager-now-works-with-napari/63982) works well, but only for a specific amount of time.

Like it slows down and stops working after a while?

ieivanov, 2022-11-19:
>In theory you can interleave pycro-manager Acquisitions or use extra axes to combine them all in one

Hm, interesting, I haven't thought about interleaved acquisitions. The two acquisitions are truly separate in that they use separate hardware, different channel configurations, different z stack parameters, and different (but synchronized - for example acq1 runs 10x faster than acq2) timing. I thought the most straightforward way to do this is through two separate MM instances, each responsible for one acquisition. But if they can be interleaved, then one MM instance should be able to do it... I can give it a try.

About Live Preview in the headless instance - a solution that would work for me would be to acquire images at 10-30 Hz, display the data in napari, and then throw it out rather than saving it. That's all possible using the `img_process_fn`. A simple widget in napari will start and stop the live preview. I just need to connect the right signals for display. I'll try that option if the inverleaved acquisition doesn't work.

Thanks for the ideas!

ieivanov, 2022-11-19:
Hm, as I'm thinking about it more - MM is currently pretty tied to the idea of one Core, one Camera, one Focus stage.

Do you think I can run two (sequenced) acquisitions with different cameras and different focus stages from the same Core at the same time?

henrypinkard, 2022-11-19:
> But if they can be interleaved, then one MM instance should be able to do it... I can give it a try.

Yeah I'm curious to know what you run into. It was designed with this as a possibility, but never tested as far as I know

> About Live Preview in the headless instance - a solution that would work for me would be to acquire images at 10-30 Hz, display the data in napari, and then throw it out rather than saving it. [...]

Yeah definitely possible--the only limit is the throughput of the ZMQ bridge. That example was made a while ago now, and both napari and PM have changed a good bit in that time. If you come up with a simpler or better way to do it, it would be great to add that into PM

> Hm, as I'm thinking about it more - MM is currently pretty tied to the idea of one Core, one Camera, one Focus stage.
> 
> Do you think I can run two (sequenced) acquisitions with different cameras and different focus stages from the same Core at the same time?

Yeah you'd need to switch the core-camera etc. so maybe this actually wouldn't be possible. But I think it wouldn't be so hard to make this configurable through new fields in acquisition events.

(btw moving this to PM because I think its more pertinent ther)