micro-manager/micro-manager #1554: Multi-D Acq failed with MultiCamera

VOndrejPM, 2022-11-11:
We have encountered an issue with MicroManager MultiCamera where the Multi-D Acq. (MDA) does not start and ends with error: Acquisition failed. This issue happens with Photometrics Prime BSI cameras as well as with the MicroManager demo cameras.

**Steps to reproduce:**
1. Set MicroManager with MultiCamera and two demo cameras. 
2. Select MultiCamera and run any MDA
3. Select Camera2 and run any MDA

**Expected result:** MicroManager will take MDA with the Camera2
**Actual result:** MicroManager throws error: Acquistion failed. 

**Notes:**
1. Camera1 works fine
2. Issue happens only after you have run MDA with MultiCamera. Before that, you can freely switch between Camera1 and Camera2 and run any MDA. 

**Log:**
```
2022-11-10T21:22:34.863320 tid12524 [IFO,App] Acquisition failed.
                                    [       ] java.lang.IndexOutOfBoundsException in Thread[clojure-agent-send-off-pool-1,6,main]
                                    [       ]   at clojure.lang.PersistentVector.arrayFor(PersistentVector.java:106)
                                    [       ]   at clojure.lang.PersistentVector.nth(PersistentVector.java:110)
                                    [       ]   at clojure.lang.RT.nth(RT.java:741)
                                    [       ]   at org.micromanager.acq_engine$tag_burst_image.invoke(acq_engine.clj:465)
                                    [       ]   at org.micromanager.acq_engine$produce_burst_images.invoke(acq_engine.clj:508)
                                    [       ]   at org.micromanager.acq_engine$collect_burst_images.invoke(acq_engine.clj:520)
                                    [       ]   at org.micromanager.acq_engine$collect.invoke(acq_engine.clj:600)
                                    [       ]   at org.micromanager.acq_engine$make_event_fns$fn__1298.invoke(acq_engine.clj:767)
                                    [       ]   at org.micromanager.acq_engine$execute.invoke(acq_engine.clj:774)
                                    [       ]   at org.micromanager.acq_engine$run_acquisition.invoke(acq_engine.clj:788)
                                    [       ]   at org.micromanager.acq_engine$run$fn__1418.invoke(acq_engine.clj:981)
                                    [       ]   at clojure.lang.AFn.run(AFn.java:24)
                                    [       ]   at java.lang.Thread.run(Thread.java:748)
```
**Environment:**
Windows 10 64bit
MicroManager 2.0.1 (20220901 | 20221012) , MicroManager 1.4 (20220926)

nicost, 2022-11-11:
Note that this only fails if the MDA is set to run sequences.  If, for instance, the MDA is run as a time lapse of 10 images with 1 s intervals (and 10ms exposures), it works.

I believe that the key problem lies in addition of the following metadata (only in burst most):

```
user:Camera-CameraChannelIndex | 0
user:Camera-CameraChannelName |  
```

These are only present under conditions that the MDA fails, and it is easy to see in the Clojure code that when it see CameraChannelIndex as 1 and it has a Vector with only one CamerChannelName in it, that it will throw an exception.  Currently digging to see what part of the code adds these tags (suspect MMCoreJ.i, but not quite sure yet).  Any help appreciated!

marktsuchida, 2022-11-11:
@nicost Those are added by `Multi Camera`:
https://github.com/micro-manager/mmCoreAndDevices/blob/8392a723178e61dca631d01377a39621e9ae7bb8/DeviceAdapters/Utilities/MultiCamera.cpp#L470-L473

It uses a (confusing) mechanism that saves the tag values in the (physical) camera device objects. Failing to clean up could explain this bug.

(The reason for using such a mechanism is presumably that it is the physical cameras that call `InsertImage` and send the images directly to the circular buffer, without Multi Camera having a chance to add its own metadata.)

nicost, 2022-11-11:
Ah, I see.  It does look like those tags need to be there.  I then believe the problem is in the AcqEng code.  Specifically:

```
(defn tag-burst-image [image burst-events camera-channel-names camera-index-tag
                       image-number-offset]
  (swap! state assoc-if-nil :burst-time-offset
         (- (elapsed-time @state)
            (core-time-from-tags (image :tags))))
  (let [cam-chan (if-let [cam-chan-str (get-in image [:tags camera-index-tag])]
                   (Long/parseLong cam-chan-str)
                   0)
        image-number (+ image-number-offset
                        (Long/parseLong (get-in image [:tags "ImageNumber"])))
        burst-event (nth burst-events image-number)
        camera-channel-name (nth camera-channel-names cam-chan)
```

It does make sen that the code throws an exception because the vector camera-channel-names has only one member, and the code asks for a second member.  A first fix would be to make sure that cam-chan is never larger than the number of camera-channel-names, but my first attempts at writing Clojure failed....  What do you think @marktsuchida?

marktsuchida, 2022-11-11:
I feel that that may hide the bug but wouldn't fix it.

Is there a reason that Multi Camera cannot restore the physical cameras to their original state so that they produce the same tags (or lack thereof) as before a Multi Camera acquisition was run? (There may be -- I haven't looked yet.)

nicost, 2022-11-12:
I don't think those tags are the culprit.  The Multi-Camera has no reliable mechanism to add them and take them away.  The Acquisition Engine should work correctly when those tags are present.  It does so when it is snapping rather than running a burst.  The section of the code in tag-burst-image should only try to set the camera-channel-name if the core camera has multiple channels.  Easy to do if that code was written in Java;)

nicost, 2022-11-12:
Fix proposed in PR #1555

marktsuchida, 2022-11-14:
I see. And the fix you propose does seem harmless. Will merge.

closed 2022-11-14 by marktsuchida