micro-manager/micro-manager #1574: Typo in user profile JSON

ziw-liu, 2022-11-23:
The user profile JSON files generated by Micro-Manager has a typo in one of its keys:

```json
            "run ZQM server": {
              "type": "BOOLEAN",
              "scalar": true
            },
```

where `ZQM` should be `ZMQ` (short for ZeroMQ).

ziw-liu, 2022-11-23:
Likely due to: 
https://github.com/micro-manager/micro-manager/blob/3b23cae1ec9e6ddf3448dac2a4a36a0293b0e53f/mmstudio/src/main/java/org/micromanager/internal/MMStudio.java#L1186-L1187

nicost, 2022-11-25:
Changing it to what it should be will change behavior for people.  Is it important to get the spelling right here?

marktsuchida, 2022-11-25:
I think in this case it is better to leave it misspelled. The alternatives would be to cause people to have to re-check the box after upgrading (this will be confusing), or to handle the old and new spelling in the code (which adds complexity), both of which seem worse than living with the misspelling.
We do appreciate the report, though, @ziw-liu.

closed 2022-11-25 by marktsuchida