micro-manager/micro-manager #1579: MultiPage Tiff storage goes balistic under certain conditions

nicost, 2022-11-30:
Disocovered by Tanner Fadero.  Steps to reproduce (beware, this may fill up your harddrive!) with the democamera:
- Set the camera size to something big, for instance: 2500x2000 (i.e. ~10MB for each 16bit image)
- Run an MDA with multiple positions (for instance: 10), 2 channels, and 200 z positions.  Make sure that Options - Save XY position in ... is checked and that the acquisition order is Position, Channel, Slice, and that the data are saved as "Image Stack File"
- Discover that instead of one file per position, there are hundreds of files in the target location, each 3.9GB.  With a little luck your hard drive will be full with unusable data.

It seems this bug is triggered by the size of one position exceeding 3.9GB.

marktsuchida, 2022-11-30:
This reminds me of the discussion we had in #1415. Could it be an edge case in the code that was changed in that PR, such that the fallback is not working?

A good way to test this, by the way, is to create a fixed-size disk image and save to it, so that your main drive doesn't fill up.

Cc: @dickinson-lab in case they have any insight.

henrypinkard, 2022-11-30:
An easy test would be to run the same thing with NDtiff and see if you get the same behavior. If not, then it is something wrong with that splitting files logic mark points out, likely

nicost, 2022-11-30:
Plot thickens a bit, this only happens with an image size of 2250 x 2550, i.e. 2500 x 2500 is OK.

dickinson-lab, 2022-11-30:
Does it happen with bigger or smaller images too, or is it only that exact size that is problematic?

Best,
Dan

Sent from my phone - please excuse short responses and typos

nicost, 2022-11-30:
Don't exactly know where the boundaries are, 2500 x 2500 is OK, I have not extensively tested more sizes, now focussing on detangling the code:

```
      if (splitByFrame) {
         size = singleImageBytes * axis1ImagesLeft
               + singleImageBytes * axis1Size * axis2ImagesLeft
               + SPACE_FOR_COMMENTS + numChannels_ * DISPLAY_SETTINGS_BYTES_PER_CHANNEL
               + extraPadding + filePosition_;
         size += omeMDLength;
```
This calculation does not seem to be correct, but will need some time to make it (more) correct.

closed 2022-11-30 by nicost

dickinson-lab, 2022-11-30:
Thanks Nico! Glad it was a quick fix.

Best,
Dan

Sent from my phone - please excuse short responses and typos

Closed #1579<https://nam12.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Fmicro-manager%2Fmicro-manager%2Fissues%2F1579&data=05%7C01%7C%7C86fb48e93f8f4bc5305508dad31811ae%7C31d7e2a5bdd8414e9e97bea998ebdfe1%7C0%7C0%7C638054397578921968%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=feCgU4FGGxplOM5F32wdrVbfJhsxPYi1vvt6GhOGy44%3D&reserved=0> as completed via #1580<https://nam12.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Fmicro-manager%2Fmicro-manager%2Fpull%2F1580&data=05%7C01%7C%7C86fb48e93f8f4bc5305508dad31811ae%7C31d7e2a5bdd8414e9e97bea998ebdfe1%7C0%7C0%7C638054397578921968%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=xlkAoW%2F3JzpmcltBqK31ntNGKWd2Q%2FO1JKxZxtP6GNU%3D&reserved=0>.

PR #1580 (2022-11-30): Studio: Fix bug in MpTiffWRiter resulting in many large files.
Calculation of the expected file size was wrong.  Classis 0 or 1 issue. Closes #1579.