micro-manager/micro-manager #1583: Java code formatting and Checkstyle

marktsuchida, 2022-12-06:
I had a Zoom chat with @henrypinkard about code formatting, style checking, pre-commit hooks, and workflow for merging PRs.

Here's a tentative proposal we came up with (with some details added by me). @nicost what do you think of this general direction? Worth trying? @henrypinkard Please feel free to elaborate, of course.

- Introduce automation so that _specific_ pull requests can be auto-merged once checks pass
    - This will be applied only to PRs that are manually marked by a core developer with a dedicated tag `automerge`
    - The intent is to only use this for trivial PRs (e.g. bumping the version of a dependency) where a core developer is sure that it should be merged as long as the tests pass but does not want to manually wait and merge
    - This should be possible with [this action](https://github.com/marketplace/actions/automatically-merge-pr) (there may be other ways)
    - This can be implemented independent of all of what follows

- Remove the pure formatting (indentation, brace position, etc.) rules from our Checkstyle configuration

- Instead, add a pre-commit hook to run clang-format (which supports Java)
    - This has the advantage that the pre-commit hook will automatically reformat files (after which you can re-stage those files and successfully commit)
    - We'll need to create a clang-format config file that is close to our current style (@marktsuchida will come up with this if we're okay with this general direction)
    - We will likely need to reformat all our Java code once (100% automatic)
    - There is an IDEA plugin for clang-format which we should test and validate (if it doesn't work, we can still provide an IDEA settings file like we do now)
    - This does not add any setup requirements for the pre-commit hook
    - We can defer our final decision on whether to make this switch until after we've actually seen how the formatting behaves

- For the other Checkstyle rules, turn them into warnings instead of errors
    - Warnings can be printed by the pre-commit hook but will not be (easily) visible on GitHub
    - Non-formatting rules include things like name capitalization, method order (overloads togeter), switch statement fall-throughs, etc.
    - These rules typically don't have automated/mechanical fixes (at least not safe ones)
    - For major changes and API changes, some of these style issues can be addressed during code review
    - (Added by @marktsuchida) We never really evaluated each of these rules and whether they help our project. It might be more productive to concentrate on the smallest subset that we truly care about (for example, do we really need to enforce comments on Unicode escapes?)

- Let's make it a rule that we don't (intentionally) merge PRs that do not pass checks (says @marktsuchida)
    - There will be no reason to merge before checks finish, once we can enable automerge for the PR
    - If there is a good reason to want to merge when checks are failing, it probably means that we should fix, improve, or disable (temporarily or permanently) the check, rather than throwing out all checking. This can be proposed in a PR or issue.

marktsuchida, 2022-12-07:
Regarding the first bullet point -- maybe GitHub's "Enable auto-merge" is sufficient for this after all. I think I misunderstood how it works earlier.

> If you allow auto-merge for pull requests in your repository, people with write permissions can configure _individual_ pull requests in the repository to merge automatically when all merge requirements are met.

(Emphasis added; GitHub [docs](https://docs.github.com/en/repositories/configuring-branches-and-merges-in-your-repository/configuring-pull-request-merges/managing-auto-merge-for-pull-requests-in-your-repository))

It does mean that we also need to make checks passing a requirement for merge (using a branch protection rule). That seems okay, especially because it looks like we can bypass the rule in emergencies.

Anyone against going ahead and enabling this?

nicost, 2022-12-11:
All looks good to me.  Fine to enable auto-merge.

henrypinkard, 2022-12-16:
Sorry for the late reply (was experimenting with the github API and accidentally turned off notifications to temporarily:) )

Thanks for the excellent writeup @marktsuchida, and the many great ideas!

I think this all sounds like a great plan. Also I tested out automerge on pycro-manager, and it its quite nice and should be a big improvement to the dependency updating workflow. I'm happy to set this up although I think I need my permissions changed on the MM organization because I cant access settings right now.

These are the settings I used. Could be nice to add the require a pull request to merge also--I have definitely accidentally pushed to the main branch before

[image]

marktsuchida, 2023-01-10:
I'm back to looking at this. It looks like I will need to either 1) let mmCoreAndDevices' auto-update action bypass the branch protection rules (not sure if this is possible for the status check requirement), or 2) change said action to create a PR (hopefully with auto-merge) instead of directly committing to `main`. I will see what I can do.

henrypinkard, 2023-01-17:
I set about to automate the version updates for PycroManager/NDTiff/NDViewer/AcqEngJ, which now seems to be working, and I think the strategies used there can be used for this as well. The [full logic](https://github.com/micro-manager/pycro-manager/tree/main/build_automation) is here, and it implements several elements you reference above @marktsuchida. Specifically:

> Introduce automation so that _specific_ pull requests can be auto-merged once checks pass
>
>     * ~This will be applied only to PRs that are manually marked by a core developer with a dedicated tag [...]

I activated automerge along with a few branch protection rules: changes to main branch must come through a PR, and all tests must pass (including the current linting, which I guess will be modified or removed based on changes here)

>     * ~This should be possible with [this action](https://github.com/marketplace/actions/automatically-merge-pr) (there may be other ways)~ see next comment

I didn't use a marketplace action, but instead used the [github command line interface directly](https://github.com/micro-manager/pycro-manager/blob/cd640d9eea6db5b67f6b0d54893f80889b9a292b/.github/workflows/maven_deploy_and_mm_update.yml#L94). (I'm actually still a bit confused about what, if any, advantages there are to using marketplace actions over just writing a bash script)

> Let's make it a rule that we don't (intentionally) merge PRs that do not pass checks (says @marktsuchida)
>
>     * There will be no reason to merge before checks finish, once we can enable automerge [...]

I agree with this, especially once your other point of turning some of currently stringent checkstyle errors into warnings is addressed. 

> It looks like I will need to either 1) let mmCoreAndDevices' auto-update action bypass the branch protection rules (not sure if this is possible for the status check requirement), or 2) change said [...]

I don't think I totally understand the issue here, but I will offer in case it was helpful the general strategy I found for PM that worked: Create a separate branch in the repo that actions can push to directly (in my case, "dependency-update"), then use the github CLI to create automerging pull requests from this branch into the main branch. This allows updates to automatically enter main, provided they pass all required status checks.

Also, @marktsuchida, a couple questions about how you set up [this action](https://github.com/micro-manager/mmCoreAndDevices/blob/main/.github/workflows/sync-micro-manager-submodule.yml) in MMCoreAndDevices: 

1. How did you generate this? `ssh-key: ${{ secrets.SSH_KEY_DEPLOY_TO_MICRO_MANAGER }}`
2. Where did this email address (`git config user.email '41898282+github-actions[bot]@users.noreply.github.com'`) come from? Is it something you generated?

marktsuchida, 2023-01-20:
Cool that you got this working!

Looking at the branch and PRs that were actually created by this, I think we need to avoid reusing the same branch for multiple PRs. It leads to hard-to-interpret commit history (take a look at `gitk` or `git log --graph` to see this). I think it is also causing the PRs to not merge cleanly, because the reused branch doesn't have the latest 'main' changes in it (leading you to manually merge 'main' into it, which results in more noise in the history).

No big deal as long as we fix it before too long. Probably the cleanest thing to do is for the action to delete and recreate the branch when the existing 'dependency-update' branch is fully merged into 'main'. Alternatively, the action can be made to always create a unique new branch, and humans can discard all but the newest if more than one accumulates before being merged.

Also, if I understand correctly, nobody can merge anything into mmCoreAndDevices right now, because it will fail to create a commit due to the branch protection rule you enabled. Or am I missing something?

> I don't think I totally understand the issue here, but I will offer in case it was helpful the general strategy I found for PM that worked: Create a separate branch in the repo that actions can push [...]

I think you did understand roughly what I had in mind. The only problem is that this needed to be applied to the mmCoreAndDevices action (which currently creates a commit directly without a PR) _before_ enabling branch protection rules. Now I need to update mmCoreAndDevices before any device code can be (successfully) merged. But you also figured out how to do the necessary fix (thanks!), so it shouldn't be hard for me to do that.

> 1\. How did you generate this? `ssh-key: ${{ secrets.SSH_KEY_DEPLOY_TO_MICRO_MANAGER }}`

Settings (of micro-manager repo) > Deploy Keys. This allows you to create as many SSH keys as necessary. Best to use a separate key for each action, so that keys can be removed when no longer needed without affecting other things.
Then add as a secret to the repo running the action (mmCoreAndDevices or pycro-manager in this case). No need to record the sensitive key anywhere else, because if you need to enter it again, it's better to create a new key and delete the old one.

> 2\. Where did this email address (`git config user.email '41898282+github-actions[bot]@users.noreply.github.com'`) come from? Is it something you generated?

[Somebody figured it out.](https://github.com/orgs/community/discussions/26560) I don't think it is important what email address we use (the nonexistent `bot@micro-manager.org` would be totally fine), but using this exact address (with the numerical prefix) displays nicely as a bot. I would only use it for commits actually created by GitHub Actions workflows, of course.

I've been a little busy this week with non-MM stuff, but will get things in order with mmCoreAndDevices early next week.

marktsuchida, 2023-01-20:
> (leading you to manually merge 'main' into it, which results in more noise in the history).

Or maybe this merge happens as part of the auto-merging process (although I find that rather yucky if that's how auto-merge works)? In any case, I think it should be avoidable if the PR doesn't reuse the branch from past PRs.

henrypinkard, 2023-01-21:
> Looking at the branch and PRs that were actually created by this, I think we need to avoid reusing the same branch for multiple PRs. It leads to hard-to-interpret commit history (take a look at [...]

Got it, that makes sense. Since I think you understand how this all works better than me, would you mind creating a PR to [the action](https://github.com/micro-manager/pycro-manager/blob/2d449c474d8d73766cc5687e78165d751e21f2e6/.github/workflows/maven_deploy_and_mm_update.yml#L51) with what you think the correct behavior should be? I'm happy to test and copy to the other pycro-manager actions after that.

> Also, if I understand correctly, nobody can merge anything into mmCoreAndDevices right now, because it will fail to create a commit due to the branch protection rule you enabled. Or am I missing something?

I just disabled the "require pull requests" part of the rule for the time being which will (hopefully) fix it for the mean time. And then we can re-enable once you have a chance to fix the mmcoreanddecives action?  

> Settings (of micro-manager repo) > Deploy Keys. This allows you to create as many SSH keys as necessary. Best to use a separate key for each action, so that keys can be removed when no longer needed [...]

Thanks! I'll implement both of these which should clean up the actions I've made nice nicely.

> I've been a little busy this week with non-MM stuff, but will get things in order with mmCoreAndDevices early next week.

Sounds good!

henrypinkard, 2023-01-23:
@marktsuchida It seems the dependency-update branch disappeared (https://github.com/micro-manager/pycro-manager/actions/runs/3982582228/jobs/6827184187). Did you perhaps delete it as an intermediate step of fixing this whole process? Or maybe there is something I don't understand with the action I wrote

marktsuchida, 2023-01-23:
Sorry, my bad - I did delete it, thinking it would be recreated next time the action creates the PR. If that was not the case, it was a mistake.

henrypinkard, 2023-01-23:
No worries, i'll recreate it for now as a temporary fix

henrypinkard, 2023-02-06:
@marktsuchida I've iterated on the automated build system for pycro-manager and dependencies so that it creates a new branch each time and deletes it after the PR has merged, which should clean up the history (let me know if you think this needs further improvement).

I also figured out some more things that could be useful in other places. One thing was the authorization tokens for being able to use the gh CLI in github actions (for e.g. opening PRs in other repos and checking their status). The ssh deploy keys are insufficient for this, as they only let you do git operations, not gh operations. I had previously used a personal access token on my account, which was not the ideal solution. The better solution was to create a [github app](https://github.com/organizations/micro-manager/settings/installations/33717628) for the micro-manager organization that has privileges for being able to do this.

I also figured out how to make an action in one repo [wait for a PR in another repo to merge](https://github.com/micro-manager/pycro-manager/blob/4008543522c95296652cd51e544dba1324fb6acf/.github/workflows/build_and_deploy.yml#L178)

In case you want to use a similar strategy on MMCoreAndDevices instead of just pushing updates to the main branch of micro-manager, I went ahead and added the keys to the repository secrets. This also means the ssh deploy keys can be deleted, since these app keys allow its privileges as well. So you should be able to pretty easily copy the [actions in pycro-manager](https://github.com/micro-manager/pycro-manager/blob/main/.github/workflows/build_and_deploy.yml) to implement. I am also happy to take a stab at this, just let me know.