micro-manager/micro-manager #1603: Micro-Manager unable to launch on MacOS 13.1 (22C65) Ventura / Apple M1 Pro

Pinaceae, 2023-01-11:
Hi there,

I use Micro-Manager on my MacBook Pro (with Apple M1 Pro chip) to view imaging data generated by a custom-built imaging system controlled by Micro-Manager.

I used to be able to run Micro-Manager 2.0.0 on MacOS Monterey (MacOS 12), following the instructions described in https://micro-manager.org/Micro-Manager_Installation_Notes . But recently after upgrading the OS to Ventura, only the ImageJ toolbar, but not the Micro-Manager main window, will be launched.

I’ve tried reinstalled it and redone the entire procedure but the issue remains.

I also tried the latest nightly build. Unlike MM 2.0.0, it did throw out some error messages concerning the incompatibility with arm64 - https://global.discourse-cdn.com/business4/uploads/imagej/original/3X/2/f/2ffe30c0973e9795210f9a7ccbbc12c44a41c3c2.png

Any idea? Thank you for any input!

marktsuchida, 2023-01-12:
Thanks. MM does not currently run an Apple M1 machines, at least not without special measures.

- MM currently uses a Java runtime environment installed separately (usually in `/Library/Java/JavaVirtualMachines`, where the JRE installer (e.g. Adoptium Temurin) places it). If this JRE is an arm64 one, it will not be able to load MMCore (the C++ module that handles device control), which is only built for Intel x86_64 currently. This is the error message you saw with the nightly build.
    - You can install an x86_64 JRE an an arm64 Mac, but I don't know if it will get used when launching MM (let us know if you try it!).
    - In any case it is not convenient to have to install an x86_64 JRE to the system, especially if you also use other Java apps.
    - So we probably need to bundle the JRE with the Micro-Manager installer to make this work cleanly (and eventually provide separate installers for x86_64 and arm64). This requires coming up with (or choosing) a new launcher to replace the current `ImageJ.app`. There are some features that depend on how the launcher currently works, so this is not a trivial change (but we need to do it).

- There might also be more stringent macOS security checks on Apple M1 Macs that prevent MM from running. If so, this can be solved by code signing (#396).

alessiamarcolini, 2023-01-26:
> You can install an x86_64 JRE an an arm64 Mac

this worked for me!! Thank you :)

marktsuchida, 2024-08-01:
See #2001.

closed 2024-08-01 by marktsuchida