micro-manager/micro-manager #1604: Suggest a microscopy to playground with micro-manager

ngxbac, 2023-01-13:
Hi, 

I have looked at https://micro-manager.org/Device_Support, however, these microscopy branches seem to be expensive to me. 
Anyone can suggest a cheaper one that works well with micro-manager within $5K of the budget? 
My purpose is to write a program that can interact with and control microscopy using micro-manager.

Thank you in advance

nicost, 2023-01-13:
Hi @ngxbac,

Sounds fun!  Can you tell a bit more about what you would like your program to do?  The nice thing with Micro-Manager is that it works with so much different hardware components.  One of the most affordable setups seems to be the RepRap (https://github.com/google/cncmicroscope-code).  It looks like this has a motorized XY and Z, as well as illuminator.  All you would need to add is a camera.  This project is 7 years old, and it may or may not work, but may give you inspiration.

henrypinkard, 2023-01-13:
@beniroquai has some awesome low cost, open source microscopes, which (maybe?) will work with MM in the not too distant future

ngxbac, 2023-01-14:
> Hi @ngxbac,
>
> Sounds fun! Can you tell a bit more about what you would like your program to do? The nice thing with Micro-Manager is that it works with so much different hardware components. One of [...]

Hi @nicost ,

I work with Image Processing. Especially, I build an AI model running on top of microscopy. It takes an input image then makes some predictions. Based on the predictions, I need to control the motorized XY and Z. Thanks for your suggestion, it is really useful to help me build some prototypes first. Really appreciate!

ngxbac, 2023-01-14:
> @beniroquai has some awesome low cost, open source microscopes, which (maybe?) will work with MM in the not too distant future

That's awesome! Thanks for sharing it @henrypinkard . It is really insightful.

beniroquai, 2023-01-14:
Hey @ngxbac; I'm a big fan of open and low-cost microscopy. You can have a look here: https://github.com/openUC2/UC2-GIT to get an idea what's possible with UC2. @henrypinkard and I discussed the topic of integrating a client (python version: https://github.com/openUC2/UC2-REST/) that talks to our firmware (https://github.com/youseetoo/uc2-esp32). I didn't have the chance to fully implement it yet. Perhaps you are keen on going the necessary steps? ;-)