micro-manager/micro-manager #1658: "Require branches to be up to date before merging" setting

marktsuchida, 2023-03-16:
In our GitHub branch protection rule for `main`, the `Require branches to be up to date before merging` setting is currently enabled.

As far as I understand, this requires that there are no new changes on `main` since the PR's branch was created. This can be satisfied by rebasing the PR branch (so that it now branches off the tip of `main`), or by merging `main` into the PR branch (which GitHub seduces you to do by clicking a button).

Unfortunately, the latter approach (merging `main`) means that whenever two or more PRs existed at the same time, all but one of them gain _nearly_ pointless merge commits that crowd out the view in `gitk` or `git log --graph` and create a lot more work when trying to understand the history of changes.

(I say "nearly" because, strictly speaking, running tests on an up-to-date branch does eliminate all chances of the tests failing after merging the PR. Without this, there is the theoretical possibility that two PRs make incompatible changes that nonetheless have diffs that do not conflict, breaking `main`. But this happens so rarely in practice (I can't even think of an example that happened to us) that I think it would be okay for `main` to break in such situations. We can deal with it.)

The other approach (rebasing onto `main`) requires manual work in the easiest of cases and may be very difficult in other cases.

I would prefer to disable this checkbox, so that pull requests that pass tests can be merged even if those tests were not run against the merge with latest `main`. In addition to cleaner history, this also has the advantage that we don't have to wait for additional tests to finish before being able to merge.

[Also, if I'm not mistaken, the tests are run against the result of (hypothetically) merging the PR into `main` at the time they are run. This means that, when in doubt, simply re-running the tests should ensure that merging with main will not result in problems (I have no idea why GitHub doesn't offer this as a convenient button to press).]

@nicost @henrypinkard What do you think?

nicost, 2023-03-16:
Sounds good to me.  I was certainly guilty of using that convenient button, and do notice that the history looks quite tangled.  Thanks for your efforts on keeping the repo and build working so well!

henrypinkard, 2023-03-16:
Sounds good to me too. I think I was the one who enabled that rule, and I don't see any downside to disabling