micro-manager/micro-manager #1661: Unable to connect to USB microscope on Ubuntu

fabiengagne, 2023-03-21:
I could not find how to make use of my 5MP USB microscope with uManager on Ubuntu 22.04 LTS.

I have a USB microscope like [this one](https://www.amazon.ca/-/fr/Microscope-num%C3%A9rique-professionnel-%C3%A9clairage-sup%C3%A9rieur/dp/B073F977CK/ref=sr_1_7?__mk_fr_CA=%C3%85M%C3%85%C5%BD%C3%95%C3%91&crid=3HHIRL22VQ2L8&keywords=5MP+Microscope+with+Stand&qid=1679364863&sprefix=5mp+microscope+with+stand%2Caps%2C115&sr=8-7).

The USB microscope is detected by the OS, and the device correctly appears in ```/dev/v4l```.

In micromanager's Device Configuration Wizard I tried to configure with the ```video4linux2/Video4Linux2: video4linux2 camera device adapter```  The configuration succeeds but live view shows the video feed of my laptop's webcam instead of the USB microscope.

marktsuchida, 2023-03-22:
I guess the question is whether your microscope works with Video4Linux at all. I would suggest finding a program that gives you better control/information on V4L devices than Micro-Manager does, and see if it can connect with your microscope.

If that works, maybe the Micro-Manager V4L device adapter needs a way to select the camera.

BTW this kind of question is best for the Image.sc forum, where there may be other users who have used V4L (as far as I know, the MM core devs have little experience with V4L).

fabiengagne, 2023-03-24:
Yes my microscope works correctly, as I can use it with OBS Studio and can even control it's colors, gain, etc.

marktsuchida, 2023-03-24:
I'm basing this just on reading the [source code](https://github.com/micro-manager/mmCoreAndDevices/blob/main/DeviceAdapters/Video4Linux/video4linux2.cpp), but have you tried setting the `DevicePath` property (default value `/dev/video0`) to something appropriate? This is in the Device Property Browser.