micro-manager/micro-manager #1663: Provide a proper installer for Linux

fabiengagne, 2023-03-21:
Provide a proper installation package for Linux through snapcraft, flatpack, or both.

henrypinkard, 2023-03-21:
Do you know how to set this up? It would be a great contribution!

fabiengagne, 2023-03-21:
Unfortunately no skills in this area.

marktsuchida, 2023-03-22:
I've been eyeing Flatpak (or possibly AppImage) as well; it seems like a possibility.
Personally I'd rather simplify our Windows/macOS installer situation before adding a third one, but let's keep this open.

JiangXL, 2023-04-17:
It will bring huge convenience if we package micromanager as snapcraft or flatpack. One of I don't know is that how their package method communicate with peripheral hardware.

JiangXL, 2023-05-10:
The flatpak of Fiji could be a reference and start point.
https://github.com/fiji/fiji-builds/pull/8

NayeliGutierrez, 2025-05-12:
Hello, any updates on the installer for Linux?
Thanks in advance!
Nayeli