micro-manager/micro-manager #1708: Graphical (interactive) view of Position List

marktsuchida, 2023-05-31:
It would be complicated to do the really nice thing of overlaying acquired images (à la DeltaVision), but perhaps not so hard to implement if we just plot the positions (as rectangles when the field size is known; spots otherwise). Ideally allow selection of position(s) linked to selection in the table, and right-click to act on a position (move-to, delete). Also display current stage position (or field of view).

Related to #1707 but perhaps useful for manually marked positions or smaller grids/plates. We were talking about something like this in our lab yesterday.

Things that might need consideration:
- Does the plot need to be simplified when the number of positions is huge? (Initial version could just refuse to render above a 
threshold.)
- Add this to Position List window? Separate window?

henrypinkard, 2023-05-31:
Much of this is implemented in Magellan/NDViewer. It would not be so hard to repackage or modify since it has largely been chopped into separate libraries with minimal dependencies. The only thing that Magellan is doing on its own at this point (i.e. not split into another library) is keeping its own concept of a position list so that it knows which positions belong to grid objects and which belong to surface objects.

Maybe it is worth considering unifying some of this functionality, and bringing explore acquisitions into mmstudio, rather than keeping them buried in the plugin

[image]

marktsuchida, 2023-06-07:
Exactly: the point is to have it available in the regular Position List.

In an ideal world, the Position List, HCS Site Generator, Magellan, etc., would not have each invented their own way of doing things. So I'm all for unifying the underlying data models, if it can be done without disrupting entrenched workflows. But the scope of this issue is just about a graphical/interactive plot (not an "explore" feature; that's something else), which I think is a much less ambitious undertaking.

henrypinkard, 2023-06-07:
Yes I think that's a good idea and should be feasible for Magellan without disruption by adding some additional metadata to the positions.

The "explore" features are a separate layer on top of that UI that shows the different positions, so that type of display can be used without having to use explore