micro-manager/micro-manager #1711: Feature request: ignore zeros in LUT autostretch function

fadero, 2023-06-03:
I'd like to request a feature be added to the micro-manager image inspector, specifically in the "Histograms and Intensity Scaling" submenu. Basically, I'd like to have an option to "Ignore Zeros" when the image autostretches the LUT. 

I'd like this because I often end up with images that have many pixels with 0 intensity (due to image transformations like deskewing OPM stacks). My argument for why this should at least be a feature is because most scientific cameras have pixel intensity offsets far above 0 to maintain sensor linearity. Therefore, zeros are more likely to arise from a processing artifact than from something physical from the sensor. In that vein, it'd be nice to be able to autostretch the image to only the nonzero values. 

Tagging @nicost here because we've discussed in person.

nicost, 2023-06-07:
Nope, this is just to ignore zeros (all of them). Ideally, the Ignore % will still work, but also not take into account any real zeros.

fadero, 2023-06-07:
It'd be nice to be able to have an option to "ignore % (including zeros)" or "ignore % (excluding zeros)"

For example, if I have the following set of pixel values:
0, 0, 0, 0, 0, 2, 4, 6, 8, 10

And I click "ignore 10% (including zeros)," then autostretch should set the min and max of my LUT to the following values:
0-8

If I click "ignore 10% (excluding zeros)," then autostretch should set the min and max of my LUT to the following values:
3-8

marktsuchida, 2023-06-07:
> Nope, this is just to ignore zeros (all of them). Ideally, the Ignore % will still work, but also not take into account any real zeros.

I think you are saying the same thing as what I meant to say, but let's see if I get this right.

> It'd be nice to be able to have an option to "ignore % (including zeros)" or "ignore % (excluding zeros)"

I find this confusing, because it is not immediately clear whether "including" means including in the "ignore" or including in the range.

Given the current UI:
☑︎ Autostretch (Ignore _____%)

Would something like the following work?
☑︎ Autostretch (Ignore _____%; ☑︎ Ignore zero pixels)

This does not make it explicit that zeros are excluded from the intensity range _before_ the percentile to ignore is computed, but I think this can be inferred (it wouldn't make any sense to do it in the opposite order).

nicost, 2023-06-07:
Agreed.  Even wonder if that ignore zero pixels can be hidden in the gear icon, since it is not something you would likely change regularly.  It also makes sense for it to be the default (except possibly for 8 bit images). 
Now hoping that implementing this is easy (if I remember correctly the histogram is calculated using imglib2).

marktsuchida, 2023-06-12:
Yeah, hiding it in the gear menu makes sense. To prevent confusion, it will be good to show an indicator when it is enabled:
☑︎ Autostretch (Ignore _____%)
☑︎ Autostretch (Ignore _____%) (Ignoring zero pixels)

I don't think it's a good idea to make it the default, though. It is not behavior that people would expect a priori, and it is something that is only advantageous when displaying processed images -- an important use case but not the most common one. Aside from 8-bit cameras, photon-counting cameras and scanning microscopes (and probably other things that I can't think of or know of) will also be affected, and I don't like the idea of the default depending on any variable if we can avoid it.

The implementation will need to modify the quantile computation, but I don't think you need to touch the histogram-computing ImgLib2 code, because it happens after the histograms have been computed (basically the new feature pretends the leftmost histogram bin contains zero, also adjusting the total pixel count accordingly). See `IntegerComponentStats.getQuantile()` and surrounding functions.

nicost, 2023-06-18:
Almost there.  Should the min value shown under the histogram also ignore zeros?

fadero, 2023-06-18:
I'd put in a weak vote to ignore the zeros in the histogram too. Normally zeros are pretty obvious in the image- I don't need a bin at 0 with a spike in the histogram to tell me that. But whatever is easier.

nicost, 2023-06-19:
I mean the values displayed for min, max, avg, stddev.  Kind of makes more sense to me for those to also ignore zeros (which creates some more work, but should be doable).

marktsuchida, 2023-06-19:
Ignoring zeros in the min/max/avg/std display seems fine to me, because those are more useful stats than the ones including zeroes in the calculation, for the intended use (= treat zeroes as missing pixels). Then the Gear menu item can be named something like "Ignore zero pixels in Autoscale and Stats", to make this clear.

The only downside is that you then have no direct way of seeing if the image (or ROI) contains any zero pixels. But unless we expect this to be a frequent need, I think the ability to switch off the ignore-zeroes feature temporarily is good enough for that. (Or there could be a 5th stat display item, showing the number of zero pixels in the image/ROI, but I don't know if that's necessary.)

closed 2023-07-01 by nicost

fadero, 2023-07-21:
@nicost I tested the "ignore zeros" feature today in the latest nightly build of MM, and the Image Inspector doesn't seem to "remember" the option to ignore zeros. I've tried to set the option in an open window (e.g. the Live window), then close the window. Image Inspector keeps that option checked until I open a new Live window, either by snapping or by starting Live again. 

Any chance you can advise on how to keep this option enabled?

PR #1725 (2023-07-01): Adds option to exclude pixels with value zero from histogram autostretch and stats.
Closes #1711