micro-manager/micro-manager #1719: Add Acquisition Settings to SummaryMetadata

nicost, 2023-06-13:
We have long wanted to have a "ReUse" button, so that one can apply the acquisition settings of a dataset to the MDA window.  Also, it would be really nice to record the acquisition settings.  It seems that just adding JSON serialized acquisition settings to the Summary Metadata would make both of these things quite easy.  Is there anything against doing this (other than that it may break when the version of the SequenceSettings changes)?

marktsuchida, 2023-06-15:
It might be good to include a version number in the `SequenceSettings` JSON, and reject any future versions when reading.

I think the complicated part (although not any different to loading sequence settings from a separate file; I'm not sure how that currently works) is what to do if the channel group, channels, or stages no longer exist. It would be nice to refuse to load (and keep all the current settings) rather than load something that doesn't work and fail after the acquisition starts.

Also the `SequenceSettings` JSON includes the directory root, name prefix, and saving format. It is not clear if the user wants to restore these when applying saved settings (maybe ask?).

One idea would be to pop up a dialog with checkboxes for each setting category (Time, Positions, Z, Channels, Saving, etc.), so that the user can decide which settings to apply. If this dialog remembers its settings, it could be useful in situations like when you want to copy the channel and exposure settings from a dataset without losing the Z range you determined for the current specimen. Not trying to make this a bigger project; just an idea.

nicost, 2023-06-15:
I think the first step is actually adding acquisition settings to the Summary Metadata (presumably in JSON format).  I'll look into adding a version number, and then open a PR before going on to ReUse.  I like the idea of that dialog for the re-use part.

PR #1740 (2023-07-20): Adds Acquisition Settings to the Summary Metadata
Reworked handling of SummaryMetadata in AcqEngJ adapter.  Added a version number to Sequence Settings, and refuse to de-serialize SequenceSettings from JSON if version is higher than the current version.  This should make it possible to "Re-Use" SequenceSettings in the near future.  Closes #1719.
closed 2023-07-20 by nicost