micro-manager/micro-manager #1723: MultipageTIFFDatastore with non-continuous position index

ieivanov, 2023-06-17:
Have an acquisition where I open a `MultipageTIFFDatastore` and manually snap images and put them in the datastore with the correct Coords and Metadata. I find that the channel and z axes can be increasing but non-continuous (e.g. [3, 5, 6]), as intended, but this does not work for the position index. I get a `Multipage Tiff storage only supports images in increasing order, t=0, t=2, etc..` error on the *second* call to `putImage`. This demo script will run with `position_idx = 0`, but not with say `position_idx = 3`. In our acquisition pipeline we do FOV selection and need to be able to save a subset of the positions in the position list manager. Any advice on how to do that? I am constrained in that the Summary Metadata needs to be written at the start and I want to keep the correct FOV position index, rather than starting at zero. Any advice will be appreciated.

```python
import os
import numpy as np
from pycromanager import Core, Studio, JavaObject

# Create datastore
data_dir = '/Users/ivan.ivanov/Documents/temp/df_acq_30'
channels = ['DAPI', 'FITC']
z_step = 1
z_positions = np.arange(0, 10, z_step)
position_idx = 0
position_name = 'Pos0'

mm_core = Core(convert_camel_case=False)
mm_studio = Studio(convert_camel_case=False)

should_split_positions = True
should_generate_separate_metadata = True
datastore = mm_studio.data().createMultipageTIFFDatastore(
    str(data_dir), should_generate_separate_metadata, should_split_positions
)

mm_studio.displays().createDisplay(datastore)

# Set summary metadata
# see https://micro-manager.org/apidoc/mmstudio/latest/org/micromanager/data/SummaryMetadata.Builder.html
smb = datastore.getSummaryMetadata().copyBuilder()

axis_order = JavaObject('java.util.ArrayList')
for axis in ('z', 'channel', 'position', 'time'):
    axis_order.add(axis)
smb = smb.axisOrder(axis_order)

smb = smb.channelGroup('Channel')
channel_names = JavaObject('java.util.ArrayList')
for channel in channels:
    channel_names.add(channel)
smb = smb.channelNames(channel_names)

smb = smb.directory(os.path.dirname(data_dir))
smb = smb.prefix(os.path.basename(data_dir))

intended_dimensions = mm_studio.data().coordsBuilder()
intended_dimensions = intended_dimensions.z(len(z_positions)).channel(len(channels)).stagePosition(1).timePoint(1)
smb = smb.intendedDimensions(intended_dimensions.build())

smb = smb.zStepUm(z_step)

pos_java_obj = JavaObject('java.util.ArrayList')
mm_position_list = mm_studio.getPositionListManager().getPositionList()
for position_ind in range(mm_position_list.getNumberOfPositions()):
    pos_java_obj.add(mm_position_list.getPosition(position_ind))
smb = smb.stagePositions(pos_java_obj)

summary_metadata = smb.build()
datastore.setSummaryMetadata(summary_metadata)

# Snap images
for z_idx, z in enumerate(z_positions):
    mm_core.snapImage()
    tmp = mm_core.getTaggedImage()
    tagged_image = mm_studio.data().convertTaggedImage(tmp)

    # see https://micro-manager.org/apidoc/mmstudio/latest/org/micromanager/data/Coords.Builder.html
    coords_builder = tagged_image.getCoords().copy()
    coords_builder = coords_builder.channel(0).z(z_idx).stagePosition(position_idx)
    coords = coords_builder.build()

    # see https://micro-manager.org/apidoc/mmstudio/latest/org/micromanager/data/Metadata.Builder.html
    metadata_builder = tagged_image.getMetadata().copyBuilderWithNewUUID()
    metadata_builder = metadata_builder.positionName(position_name)
    metadata_builder = metadata_builder.xPositionUm(0).yPositionUm(0).zPositionUm(0)
    metadata_builder = metadata_builder.exposureMs(10)
    metadata = metadata_builder.build()

    tagged_image = tagged_image.copyWith(coords, metadata)
    datastore.putImage(tagged_image)

# Save datastore
datastore.freeze()
# datastore.close()
```

marktsuchida, 2023-06-19:
Does it matter if you set the intended dimensions to include the maximum possible position index?

ieivanov, 2023-06-19:
Hi Mark! Sorry, this example contains a bug, as you point you. In my code I set the intended dimension to the maximum possible position index and I still get the error I described.