micro-manager/micro-manager #1726: svn connection timed out

leanedonze, 2023-06-26:
Hi there,

Thanks a lot for the software!

I'm trying to install it on my Jetson Orin Nano running Ubuntu 20.04 following these instructions https://micro-manager.org/Linux_installation_from_source_MM2 

However, when I run 
`svn co https://valelab4.ucsf.edu/svn/3rdpartypublic/`

I get the following error 
`svn: E170013: Unable to connect to a repository at URL 'https://valelab4.ucsf.edu/svn/3rdpartypublic/'
`
`
svn: E000110: Error running context: Connection timed out
`

I tried on another computer as well, so I believe this is not related to my internet connection.

Any idea how I can solve this issue?

Thank you for your help

henrypinkard, 2023-06-26:
It seems like valelab4.ucsf.edu may be down. @nicost @marktsuchida any ideas about this?

nicost, 2023-06-26:
Yes, looks like Peter finally pulled the plug.  We really need to host this this outside of UCSF...

nicost, 2023-06-27:
The repository is back up again.  Power outage at UCSF....

closed 2023-06-27 by nicost

leanedonze, 2023-06-28:
Oh, seems like I just tried to install it at the wrong time haha

I confirm I can run the command now, thanks!