micro-manager/micro-manager #1731: Cannot populate metadata's userData in Beanshell scripts

kmdouglass, 2023-06-30:
Hello!

I've been really struggling to perform a seemingly straightforward task for the past two days.  After trying many things, I suspect there might be a bug.

In short, I would like to populate the userData `PropertyMap` of a `Metadata` instance with some custom acquisition-related data in a Beanshell script. However, I cannot seem to construct the `PropertyMap` correctly because passing it to the `Metadata.Builder`'s `userData` method always fails.

Here's a minimal example in which I *think* I avoided all deprecated classes and methods according to the API docs at https://micro-manager.org/apidoc/mmstudio/latest/:

```
import org.micromanager.data.Datastore;
import org.micromanager.display.DisplayWindow;
import org.micromanager.PropertyMap;

Datastore store = mm.data().createRAMDatastore();
DisplayWindow display = mm.displays().createDisplay(store);
mm.displays().manage(store);
cb = mm.data().coordsBuilder();

images = mm.live().snap(true); 
img = images.get(0);

metadata = img.getMetadata();
userData = metadata.getUserData().copyBuilder();
userData.putInteger("foo", 1).build();
metadata.copyBuilderPreservingUUID().userData(userData).build();
coords = cb.time(i).build();
cImg = img.copyWith(cb.time(i).build(), metadata);
store.putImage(cImg);

store.freeze();
```

Running the above Beanshell script in Micro-Manager 2.0.1 20230619 with the default demo-camera configuration produces the following error:

```
Line 17: evaluation error : Sourced file: inline evaluation of: ``import org.micromanager.data.Datastore; import org.micromanager.display.DisplayW . . . '' : Error in method invocation: Method userData( org.micromanager.internal.propertymap.DefaultPropertyMap$Builder ) not found in class'org.micromanager.data.internal.DefaultMetadata$Builder'
```

I also attempted to use the `PropertyMaps` class (note the 's' at the end) but failed with the same result as reported here: https://forum.image.sc/t/add-userdata-to-micro-manager-metadata-in-a-beanshell-script/83080

Any ideas about what's going on here?

Thanks for the help!

nicost, 2023-07-01:
Hi Kyle, it took me a while to wrap my head about the builder concept, but it seems to be good programming practice.  The following script works for me (not extensively tested though):

```
import org.micromanager.data.Datastore;
import org.micromanager.display.DisplayWindow;

Datastore store = mm.data().createRAMDatastore();
DisplayWindow display = mm.displays().createDisplay(store);
mm.displays().manage(store);
cb = mm.data().coordsBuilder();

images = mm.live().snap(true); 
img = images.get(0);

metadata = img.getMetadata();
userDataBuilder = metadata.getUserData().copyBuilder();
userData = userDataBuilder.putInteger("foo", 1).build();
newMetadata = metadata.copyBuilderPreservingUUID().userData(userData).build();
coords = cb.time(1).build();
cImg = img.copyWith(coords, newMetadata);
store.putImage(cImg);

store.freeze();
```

The issue in your code was that you "build" the builder, but did not assign the new object to a variable.  I hope the naming of variables in this script here makes things a bit more clear, but please do ask again before spending lots of time.

kmdouglass, 2023-07-01:
Ah, thanks Nico! You're exactly right; I guess I just needed another pair of eyes on this.

I'm closing the issue. Sorry for the noise.

closed 2023-07-01 by kmdouglass