micro-manager/micro-manager #1739: PositionList in SummaryMetadata

nicost, 2023-07-06:
Currently, the following Clojure acquisition engine code translates a PositionList into JSON:

```
(defn summarize-position-list [position-list]
  (let [positions (seq (.getPositions position-list))]
    (JSONArray.
      (for [msp positions
            :let [label (.getLabel msp)
                  grid-row (.getGridRow msp)
                  grid-col (.getGridColumn msp)
                  default-xy (.getDefaultXYStage msp)
                  default-z(.getDefaultZStage msp)
                  device-positions (:axes (MultiStagePosition-to-map msp))]]
        (let [json-positions (JSONObject.
                               (into {}
                                     (for [[device coords] device-positions]
                                       [device (JSONArray. coords)])))]
          (JSONObject. {"Label" label
                        "GridRowIndex" grid-row
                        "GridColumnIndex" grid-col
                        "DefaultXYStage" default-xy
                        "DefaultZStage" default-z
                        "DeviceCoordinatesUm" json-positions}))))))
```

The JSON generated by this code is inserted in the (JSON) SummaryMetadata under the key ```InitialPositionList```.   That JSON is handled by the following code in the class ```PropertyKey```:

```
 STAGE_POSITIONS("StagePositions", "InitialPositionList", "POSITIONS",
         SummaryMetadata.class, PositionList.class) {
      @Override
      protected void convertFromGson(JsonElement je, PropertyMap.Builder dest) {
         List<PropertyMap> positions = Lists.newArrayList();
         for (JsonElement e : je.getAsJsonArray()) {
            positions.add(NonPropertyMapJSONFormats.multiStagePosition().fromGson(e));
         }
         dest.putPropertyMapList(key(), positions);
      }

      @Override
      public JsonElement convertToGson(PropertyMap source) {
         if (!source.containsPropertyMapList(key())) {
            return null;
         }
         JsonArray ja = new JsonArray();
         for (PropertyMap msp : source.getPropertyMapList(key())) {
            ja.add(NonPropertyMapJSONFormats.multiStagePosition().toGson(msp));
         }
         return ja;
      }
   },
```

It is unclear to me whether (and how) I can use the PropertyKey class to carry out the conversion from a PositionList to the JSON to be added to SummaryMetadata to store the PositionList.  I need to do this for the AcqEngJ adapter.  I could just rewrite the Clojure code to Java, but that seems error prone and cause code duplication.  The code seems to be in the ```convertToGson``` method in ```STAGE_POSITIONS```, but I am not allowed to use that since its parent is declared protected.  How do I do this?

Related, when I add SequenceSettings to the Summary Metadata, how do I do that following the PropertyKey approach?

nicost, 2023-07-06:
I even copied the convertToGson function and used that, but parsing the output generated by that function gives the error: "This is not a JSON Array".

marktsuchida, 2023-07-06:
The protected methods of `PropertyKey` provide the per-key implementation for serialization/deserialization, where special treatment is needed. The way to use it from outside code is to call `storeInGsonObject()` and `extractFromGsonObject()`, defined near the end of `PropertyKey`, which allow serialization/deserialization of a whole property map in one go. Searching for usage of these functions hopefully will provide an idea of how to use them.

The code to store/retrieve the position list into a property map is in `DefaultSummaryMetadata` (see `getStagePositionList()` and corresponding builder methods `stagePositions()`).

The `PositionList` class also has a `toPropertyMap()`, but is missing a `fromPropertyMap()`. This is because `PositionList`, being a legacy class, is mutable and doesn't conform to the design of the other 2.0 data objects (and we need to modify it without replacing the whole object). But it does have a `replaceWithPropertyMap()`, which, if I'm not mistaken, might be able to directly process a property map obtained from `SummaryMetadata`, since the same keys are used.

So, hopefully you don't need to call or modify `PropertyKey` directly if the goal is to transfer stage positions between `SummaryMetadata` and `PositionList`. If this is not the case, then probably some of the methods named above need fixing.

The idea is that within MMStudio, it is preferable to pass metadata around as property maps rather than any JSON or Gson representation. If converting to some other format/type, it logically makes sense to go from `PositionList` or `SummaryMetadata` directly, but you could also read it from a plain property map by imitating the implementations of those 2 classes (looping over `MultiStagePosition` objects).

The compatibility between acqEngine (Clojure) and MMStudio is purely at the level of the JSON serialization (which is also baked into our file format and therefore should not be changed without great care).

nicost, 2023-07-06:
I am very uncertain about all of this, and highly confused.  AcqEngJ tries to emulate acqEngine, and in its current version provides SummaryMetadata as a JSON String (just as the Clojure engine).  Should we not be doing that, and construct SummaryMetadata directly?  That would need extensive refactoring of MMAcquisition.  

In any case, a round trip of the position list, such as in:
   ```
 summaryMetadata.put(PropertyKey.STAGE_POSITIONS.key(),
                 studio_.positions().getPositionList().toPropertyMap().toJSON()); 
```
followed by:
 ```
summary = DefaultSummaryMetadata.fromPropertyMap(
                 NonPropertyMapJSONFormats.summaryMetadata().fromJSON(
                         summaryMetadata.toString()));
```
does not work, and I have tried many, many variants.

nicost, 2023-07-08:
I have taken out as much as possible of the JSON version of SummaryMetadata in AcqEngJ.  It involved re-writing part of MMAcquisition.  I do believe this is better, and could even do the same for the old acquisition engine.  Have a look at PR #1740

closed 2023-07-08 by nicost