micro-manager/micro-manager #1743: Access to AcquisitionEngine

wl-stepp, 2023-07-20:
Dear Micro-Manager team,

One of my plugins wants to know if changes are made in the MDA window. I have been using the AcquisitionEngine to add a SettingsListener to get events for changes in the settings.
`studio.getAcquisitionEngine().addSettingsListener(this);`

The getAcquisitionEngine funcionality seems to have been removed from the API completely. It would be great to have another way to receive changes in the MDA window as they happen.

See image.sc post below.
Thanks,
Willi

Image.sc discussion:
https://forum.image.sc/t/stopping-acquisition-programmatically/79056/3

closed 2023-08-26 by nicost

nicost, 2023-08-26:
With the next nightly build, you should be able to subscribe to the studio event bus:

`studio_.events().registerForEvents(this);`

Then, when the acquisition settings in the acquisition engine change (changes in the MDA window are immediately send to the engine), will result in the event bus calling you implementation of:

```
@Subscribe
public void onSettingsChanged(AcquisitionSettingsChangedEvent event) {
   SequenceSettings newSettings = settings.getNewSettings();
}
```

wl-stepp, 2023-08-31:
Awesome, thank you. Will test next week,

PR #1774 (2023-08-26): Replace AcqListeners with AcquisitionSettingsChangeEvent in API.
This simplifies the code, and enables external code to listen for changes in the MDA window.
Also avoid updating the GUI twice when a setting changes.
Bumps version to 2.0.3 since there is an API addition.
Closes #1743