micro-manager/micro-manager #1744: Image width and height missing from Metadata

nicost, 2023-07-21:
Nenad and I realized that SummaryMetadata do not contain information about image width and height.  The Clojure acquisition engine supplies that information, but it is not converted to SummaryMetadata (or it sometimes can be found as both 0 is the SummaryMetadata,txt file).  This causes several problems, one of them that to determine image size, images actually need to be opened, rather than reading a text file containing the Summary Metadata, the other that it would be really useful for ImageProcessors (which start of with SummaryMetadata) to set things up knowing what image sizes to expect.  

I realize that it is nice for future compatibility not to rely on image size information in SummaryMetadata.  If that is indeed the case, we should at least make it an obligatory element of Image Metadata (where I also often can not find it).  Even though this information is internally present though the Image object, it would be very useful for outside readers to be able to extract this information from the metadata files.

Let me know if it is OK to add this to the SummaryMetadata, and enforce that all code paths actually add it.

marktsuchida, 2023-07-25:
I don't remember, but I suppose this must be a result of trying to have a single (or fewer) sources of truth. Not against including it in the summary metadata if the datastores then reject any attempt to add images of a different size. If we leave any chance of the two not matching when writing, then it could cause quite a mess.

nicost, 2023-08-02:
It turns out that the Clojure acquisition engine adds that data, but it is removed in the step:

```
SummaryMetadata summary = DefaultSummaryMetadata.fromPropertyMap(
                  NonPropertyMapJSONFormats.summaryMetadata().fromJSON(
                           summaryMetadata.toString()));
```

but they are dropped in `NonPropertyMapJSONFormats.SummaryFormat`

closed 2023-08-07 by nicost