micro-manager/micro-manager #1746: Micro-manager is saving image data of the wrong size

ieivanov, 2023-07-24:
If I use the demo config, snap an image and save it using the Save button on the preview window (as image stack file) I get a .tif file of 682 KB, which largely makes sense.

If I change the `Camera` `OnCameraCCDXSize` and `OnCameraCCDYSize` to 64 pixels I get a .tif file of 10.2 MB, which is too large - the image should be closer to 8 KB.

Cropping the image also leads to data of the wrong size. If I restart Micro-manager and crop the image to quarter size by pressing the "Set ROI to center quad of camera" button (i.e. 256x256) I get a file size of 778 KB, when the image should be closer to 128 KB.

This wrong file size carries over to MDA acquisitions as well. I believe this bug is leading to https://github.com/micro-manager/pycro-manager/issues/639 and https://github.com/micro-manager/pycro-manager/issues/620

@nicost @henrypinkard

henrypinkard, 2023-07-25:
Nice detective work

> This wrong file size carries over to MDA acquisitions as well. I believe this bug is leading to [micro-manager/pycro-manager#639](https://github.com/micro-manager/pycro-manager/issues/639) and [micro-manager/pycro-manager#620](https://github.com/micro-manager/pycro-manager/issues/620)

It's hard to guess what common component there is that might be causing this. Pycromanager acquisition is AcqEngJ + NDTiff, where as snap and save is no acq engine + OME-Tiff. I would say its probably the demo camera, but those issues are related to physical hardware. That leaves the core, but its hard to imagine that there could be an undiagnosed problem with image size here...

So perhaps this is multiple bugs?

marktsuchida, 2023-07-25:
Just to check, are you referring to the fact that the TIFF file contains wasted space, not any data integrity issue?

There is a 4D "IndexMap" (@henrypinkard's invention from back in 2014) that maps T-P-Z-C coordinates to IFD offsets (so that image planes can be retrieved without following all IFDs through the file, which is very time-consuming). This index is preallocated at the beginning of the file, but its size is bigger when the images are smaller, because it anticipates that more images can fit within the 4 GiB limit. See `indexMapSpace` in `writeMMHeaderAndSummaryMD()` in `mmstudio/src/main/java/org/micromanager/data/internal/multipagetiff/MultipageTiffWriter.java`.

I found an old (non-public) email thread in which this issue was discussed in 2015 with the following quote:
```
128x128: 2.5MB
64x64: 10MB
32x32: 40MB
16x16: 160MB
```
This is indeed problematic, but not unexpected from how the code works. I guess we've gotten away with it because it is relatively rare for such small images to be saved individually (as opposed to in a relatively large time series). It can be said that the Multipage Tiff format, as it currently stands, is not suitable for small numbers of small images, or any number of very small images.

To fundamentally fix this, we need to write the index map at the end of the file, after truncating it to the necessary length; this could be done by storing the in-progress index map in a separate temporary file and/or in memory. Plus, we need a way to indicate in the file header that the index map is not located at the fix offset it is currently located at -- hopefully this is possible because the index map's start offset is a field at a fixed offset from the beginning of the file.

henrypinkard, 2023-07-26:
Good memory @marktsuchida! I had forgotton about this. Though the rationale is now coming back to me. The index map was originally stored in memory temporarily and and then written at the end of the file when the file filled up or acquisition finished, but after having a few datasets that crashed during writing and lost it (necessitating a bunch of code for "fixing" the index after the fact), I switched it to the beginning. But this comes with the disadvantage you point out. This is what lead me with NDTiff to ditch having the index map in the tiff file after all and just write it as its own file in real time, since it avoids a lot complexity.

ieivanov, 2023-07-26:
Thank you both for the context here. Mark, you're correct, I'm pointing out a problem with wasted space rather than a data integrity issue. It's good to here that this was a design decision and not a bug in the code. I think your approach makes sense to me and from my point of view doesn't need to be changed. You're welcome to close this issue or keep it open for future reference.