micro-manager/micro-manager #1784: Micro-Manager can write NDTiff, but not read them.

nicost, 2023-09-07:
Choosing NDTiff in the MDA window produces files, but using the File Open menu item on these files (or drag and drop into the main window) does nothing.

nicost, 2023-09-07:
2023-09-06T20:26:38.681187 tid23816 [IFO,App] Thread 95 (Thread-13) terminated with uncaught exception
2023-09-06T20:26:38.681222 tid23816 [IFO,App] java.lang.RuntimeException: java.lang.NullPointerException
2023-09-06T20:26:38.681910 tid23816 [IFO,App]   at org.micromanager.data.internal.ndtiff.NDTiffAdapter.addEssentialImageMetadata(NDTiffAdapter.java:347)
2023-09-06T20:26:38.681916 tid23816 [IFO,App]   at org.micromanager.data.internal.ndtiff.NDTiffAdapter.getAnyImage(NDTiffAdapter.java:206)
2023-09-06T20:26:38.681919 tid23816 [IFO,App]   at org.micromanager.data.internal.DefaultDatastore.getAnyImage(DefaultDatastore.java:186)
2023-09-06T20:26:38.681922 tid23816 [IFO,App]   at org.micromanager.data.internal.DefaultDataManager.loadData(DefaultDataManager.java:242)
2023-09-06T20:26:38.681924 tid23816 [IFO,App]   at org.micromanager.data.internal.DefaultDataManager.promptForDataToLoad(DefaultDataManager.java:203)
2023-09-06T20:26:38.681928 tid23816 [IFO,App]   at org.micromanager.internal.menus.FileMenu.lambda$promptToOpenFile$4(FileMenu.java:104)
2023-09-06T20:26:38.681931 tid23816 [IFO,App]   at java.lang.Thread.run(Thread.java:748)

henrypinkard, 2023-09-07:
Fixed by https://github.com/micro-manager/NDTiffStorage/pull/134

closed 2023-09-07 by henrypinkard