micro-manager/micro-manager #1791: New Micro-manager versioning scheme

nicost, 2023-09-13:
So far, we have equated the version of MMStudio with the version of the Micro-Manager release.  @marktsuchida proposed to uncouple the two and switch to a simplified date for the release, such as 2023-1, possibly coupled with a yearly or twice a year release cycle.  This would make it possible to change the major version of MMStudio according to semver, whenever we feel it is needed without building an official release.  

I like this idea very much, and believe we can switch without too much effort.  Couple of obvious questions:
- Do we want to stick to a once or twice a year release schedule?  Should there be a fixed date, by which we freeze new features, test for a few weeks, then release?
- Should the name be 2023-1, 2023-2, or 2023A, 2023B, or something completely different?

marktsuchida, 2023-10-23:
Discussed in 2023-10-23 core dev meeting and approved. Details to be worked out.

marktsuchida, 2023-11-01:
Turns out I had already written this up as #1301. [CalVer](https://calver.org/) provides some case studies on the naming scheme. I think I like 2023.11, 2024.5 (using the month), but it might be good to have a plan for when we need two in the same month -- maybe 2023.11.0, 2023.11.1, with anything other than .0 being rare.

I'd say we don't need to commit to a particular release interval (always can later if we think it's a good idea). Also probably easier to avoid feature freezes and just release a version that we deem stable (which need not be the tip of main at the time of release). Alternatively, we could simply promote a chosen nightly build to "release" status by overwriting its version info.

Rough tasks needed once we decide on naming specifics:
- Change MMStudio to use its own version number rather than getting it from version.txt (plus build scripts) during the build (eventually this will be in a pom.xml, but for now it can be in MMVersion.java)
- Add a package-wide version file (probably JSON), generated during the build and installed by the installer
- Make MMStudio display the package-wide version info (when available) where it shows the version (title bar; About dialog)
- Make sure MMStudio APIs that return the version do the right thing (probably should check where it is used currently)

marktsuchida, 2025-06-18:
Also note: the 2.x version number can then be repurposed for the MMStudio API version (SemVer).