micro-manager/micro-manager #1800: metadata specification

tlambert03, 2023-09-24:
Hi folks,

I'm at a place where I'm composing metadata to save as it comes off the scope, and I'm looking through the various ways this has been done in the past.  I know you all have done this a million times, so I apologize if there are established guides/specs/standards somewhere, but I failed to find it so far.  It looks more or less like each attempt has established its own internal set of keys. 
Here are various sets of metadata and keys that I've found:

- [summary metadata as output by the clojure engine](https://github.com/micro-manager/micro-manager/blob/main/acqEngine/src/main/clj/org/micromanager/acq_engine.clj#L924-L974)
- [summary metadata as output by AcgEnJ](https://github.com/micro-manager/AcqEngJ/blob/5f8e0a33c75be7f05c6bda55a327aac3a7f1ebf0/src/main/java/org/micromanager/acqj/main/AcqEngMetadata.java#L181-L209)
- [summary metadata builder in micro-manager](https://github.com/micro-manager/micro-manager/blob/be3f36583ebf263552146adfb8b9cdac455005ad/mmstudio/src/main/java/org/micromanager/data/SummaryMetadata.java#L41)
- keys defined by clojure [here](https://github.com/micro-manager/micro-manager/blob/be3f36583ebf263552146adfb8b9cdac455005ad/acqEngine/src/main/clj/org/micromanager/acq_engine.clj#L924-L974) and [here for frames](https://github.com/micro-manager/micro-manager/blob/be3f36583ebf263552146adfb8b9cdac455005ad/acqEngine/src/main/clj/org/micromanager/acq_engine.clj#L120-L149)
- keys defined by AcgEnJ [here](https://github.com/micro-manager/AcqEngJ/blob/5f8e0a33c75be7f05c6bda55a327aac3a7f1ebf0/src/main/java/org/micromanager/acqj/main/AcqEngMetadata.java#L40-L86)
- many of these keywords have correlates [in MMCore itself](https://github.com/micro-manager/mmCoreAndDevices/blob/a9e69215f6f6aded2c76cad69f0919dc84d121fa/MMDevice/MMDeviceConstants.h#L87-L142)

if I'm not mistaken, it appears that what one might find in either the frame metadata and/or the summary data is currently very much dependent on the acquisition engine.  Is that that case?  
- Are there any documents anywhere defining anything approaching a name/type convention or schema for what format this metadata should take?  
- Is there any benefit in trying to conform to any particular format, or is it still largely internal consumption by each tool?  If not, is this something that you feel would be beneficial to coordinate on?  Or is it something that you think is more trouble than it's worth?

mostly, I'm trying to determine if pymmcore-plus should be making at attempt to output metadata matching these formats from micro-manager or AcgEngJ.  For example, I do have a [`getTags()`](https://github.com/pymmcore-plus/pymmcore-plus/blob/bf42f430dc55f70d781905f42d50455854a9be26/src/pymmcore_plus/core/_mmcore_plus.py#L1468) method that matches the output of MMCoreJ's tags in TaggedImage, but it's not entirely clear to me whether It *matters* whether those keys match or not.  (If not, I'd be inclined to reconsider stuff and declare yet another spec... or simply always normalize to the OME data model, using [ome-types](https://github.com/tlambert03/ome-types) for example).

thoughts?

nicost, 2023-09-25:
As you have experienced directly, during the history of Micro-Manager, various keys have been used to denote the same thing.  @marktsuchida attempted to clean this up and - while staying backwards compatible - unify the keywords by running them all through the class: https://github.com/micro-manager/micro-manager/blob/main/mmstudio/src/main/java/org/micromanager/data/internal/PropertyKey.java.  All keys generated elsewhere are (supposed to be) normalized in this class.  As far as I know, this takes place for both the Clojure and Java engines, so I expect the result to be identical (or at least, very close), even though there are no tests in place to guarantee this (adding those tests is a great idea). 

To be honest, this class has always been a bit of a mystery to me,  It is very difficult to debug, and there often are subtly issues, but it surely is a daring attempt to normalize the metadata generated by various pieces of Micro-Manager code.

tlambert03, 2023-09-25:
Thanks @nicost!

> but it surely is a daring attempt to normalize the metadata generated by various pieces of Micro-Manager code.

😂 ok
I think I won't worry about it then?  Will just write OME meta

nicost, 2023-09-25:
I should probably have been a bit more positive.  The classes SummaryMetadata and ImageMetadata describe the "Micro-Manager Metadata". That PropertyKey class does a heroic job to translate all the variations that circulate in the real world (including old Micro-Manager versions) to these classes.  So, I believe that if you would create an implementation of those classes, you would be in a great state to save data that can also be read by the Java Micro-Manager version.

You may not care about that aspect.  In that case, you can choose anything you like.  When we last looked into the OME metadata (a long time ago), it had extensive options for all gasses that could go into lasers, but nothing for pixelsize.  It may be a bit better nowadays.

In related news, Glencoe will put some efforts into improving Bioformats with respect to Micro-Manager data sets.

tlambert03, 2023-09-25:
yep, they have [pixelsize](https://www.openmicroscopy.org/Schemas/Documentation/Generated/OME-2016-06/ome_xsd.html#Pixels_PhysicalSizeX).  thanks again @nicost

tlambert03, 2023-09-25:
will close this issue, but feel free to reopen if you think there's any more discussion to be had

closed 2023-09-25 by tlambert03