micro-manager/micro-manager #1807: Micro-Magellan broken

nicost, 2023-10-03:
When running an acquisition in Micro-Magellan with multiple channels, the down-sampled view is broken.  When I zoom in, things look OK again.  Problem is best illustrated with a screenshot, but I do not see an obvious way to add that here.

henrypinkard, 2023-10-05:
This doesn't reproduce for me with the latest nightly build. Are you working on the latest nightly build?

If so, one thing that would be good to know for narrowing this down is whether it is a problem with writing low res data or reading it. If you open the data from disk, does it look as expected? Or if you drag the row low res tiffs onto ImageJ. is the data there?

nicost, 2023-10-05:
Reproduced when opening the data. Sorry, do not have the bandwitdh right now to post data. Yes, tgis was on a very recent build (may have been 9/12).

nicost, 2023-10-05:
Created a 3x3 grid, and selected 3 channels (DAPI, FITC, TRITC), pressed Run Acquisition.  Image look completely wrong.

[image]

However, when zooming in one level, everything falls into place:

[image]

And yes, things look exactly the same when opening the saved dataset. Not quite sure how to get that data to you, it is 396MB.

henrypinkard, 2023-10-05:
This means that something is wrong with the higher resolution data (or the reading of it). I sent you a google drive link for transfer

nicost, 2023-10-05:
Data was uploaded

henrypinkard, 2023-10-05:
Thanks!

Okay so the data is missing at the higher resolution levels, which means something broke during the writing of higher resolution data. And it didn't happen for all images, which maybe suggests a race condition or a file system error specifically when trying to write that data. But I'm not able to reproduce this.

Can you reproduce this if you acquire data again? Anything suspicious in the core log?

nicost, 2023-10-05:
Will try tomorrow (and check the logs for that session). 

Since the lower resolution data is OK, it seems that the correct data was acquired, and correctly converted to lower resolution.  

B.t.w., how do you check this data, i.e. how do you open it other than through Micro-Magellan?

nicost, 2023-10-06:
Possibly a related question, why is not possible to zoom out further?  I am never able to show all the acquired tiles at one time because the viewer does not zoom out enough.

henrypinkard, 2023-10-11:
> Will try tomorrow (and check the logs for that session).
> 
> Since the lower resolution data is OK, it seems that the correct data was acquired, and correctly converted to lower resolution.

I mispoke there--I the _lower resolution_ (higher downsampling) data is missing. So something went wrong in the downsampling process.

> 
> B.t.w., how do you check this data, i.e. how do you open it other than through Micro-Magellan?

You can do it [through python ](https://pycro-manager.readthedocs.io/en/latest/read_data.html#), but I was just dragging the raw tiff files in different downsample levels onto imageJ

I think that exception is because the viewer is getting zeros data when it expects something. Related, but very unlikely to be causing it.

> Possibly a related question, why is not possible to zoom out further? I am never able to show all the acquired tiles at one time because the viewer does not zoom out enough.

Im guessing because the lower resolution data that it would need to load is missing?

I think this will need to be investigated by setting a break point and stepping through the data saving code in NDTiffStorage. Here is probably where I would start looking:

https://github.com/micro-manager/NDTiffStorage/blob/3fdf4ec7ec0e2ab04b0968d89b9644e6b9c44c59/java/src/main/java/org/micromanager/ndtiffstorage/NDTiffStorage.java#L927

this is after the high resolution image has saved, but before the low res one added.

If this is some kind of race condition, adding a pause there could fix it.

I'm happy to try to fix it if you can narrow down where the problem is coming from

nicost, 2023-10-12:
Setting a breakpoint there makes no difference (i.e., same problem in the viewer/data, no useful output in the logs).  Setting a breakpoint in org.micromanager.ndtifstorage.NDTiffStorage line 709 results in the following errors (repeated many, many time).  I have no idea how to go about soling this.,

java.lang.RuntimeException: Error reading image metadata from file
	at org.micromanager.ndtiffstorage.NDTiffReader.readTaggedImage(NDTiffReader.java:307)
	at org.micromanager.ndtiffstorage.NDTiffReader.readImage(NDTiffReader.java:173)
	at org.micromanager.ndtiffstorage.ResolutionLevel.getImage(ResolutionLevel.java:180)
	at org.micromanager.ndtiffstorage.NDTiffStorage.addToLowResStorage(NDTiffStorage.java:720)
	at org.micromanager.ndtiffstorage.NDTiffStorage.populateNewResolutionLevel(NDTiffStorage.java:674)
	at org.micromanager.ndtiffstorage.NDTiffStorage.increaseMaxResolutionLevel(NDTiffStorage.java:692)
	at org.micromanager.magellan.internal.magellanacq.MagellanAcqUIAndStorage.increaseMaxResolutionLevel(MagellanAcqUIAndStorage.java:338)
	at org.micromanager.ndviewer.internal.gui.DataViewCoords.updateResIndex(DataViewCoords.java:118)
	at org.micromanager.ndviewer.internal.gui.DataViewCoords.setFullResSourceDataSize(DataViewCoords.java:86)
	at org.micromanager.ndviewer.internal.gui.DisplayModel.zoom(DisplayModel.java:136)
	at org.micromanager.ndviewer.internal.gui.DisplayModel.updateDisplayBounds(DisplayModel.java:348)
	at org.micromanager.ndviewer.main.NDViewer.newImageArrived(NDViewer.java:221)
	at org.micromanager.magellan.internal.magellanacq.MagellanAcqUIAndStorage$3.run(MagellanAcqUIAndStorage.java:201)
	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)
	at java.util.concurrent.FutureTask.run$$$capture(FutureTask.java:266)
	at java.util.concurrent.FutureTask.run(FutureTask.java)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
	at java.lang.Thread.run(Thread.java:748)
2023-10-12T12:14:09.351536 tid3368 [IFO,dev:CSUW1-SoraMagnifier] Get FRAP answer is: 2:A
2023-10-12T12:14:10.298066 tid3368 [IFO,dev:CSUW1-SoraMagnifier] Get FRAP answer is: 2:A
java.util.concurrent.ExecutionException: java.lang.RuntimeException: java.lang.RuntimeException: Error reading image metadata from file
	at java.util.concurrent.FutureTask.report(FutureTask.java:122)
	at java.util.concurrent.FutureTask.get(FutureTask.java:192)
	at org.micromanager.magellan.internal.magellanacq.MagellanAcqUIAndStorage$3.run(MagellanAcqUIAndStorage.java:193)
	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)
	at java.util.concurrent.FutureTask.run$$$capture(FutureTask.java:266)
	at java.util.concurrent.FutureTask.run(FutureTask.java)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.RuntimeException: java.lang.RuntimeException: Error reading image metadata from file
	at org.micromanager.ndtiffstorage.NDTiffStorage$5.call(NDTiffStorage.java:1020)
	at org.micromanager.ndtiffstorage.NDTiffStorage$5.call(NDTiffStorage.java:1014)
	... 5 more
Caused by: java.lang.RuntimeException: Error reading image metadata from file
	at org.micromanager.ndtiffstorage.NDTiffReader.readTaggedImage(NDTiffReader.java:307)
	at org.micromanager.ndtiffstorage.NDTiffReader.readImage(NDTiffReader.java:173)
	at org.micromanager.ndtiffstorage.ResolutionLevel.getImage(ResolutionLevel.java:180)
	at org.micromanager.ndtiffstorage.NDTiffStorage.addToLowResStorage(NDTiffStorage.java:720)
	at org.micromanager.ndtiffstorage.NDTiffStorage.access$1400(NDTiffStorage.java:51)
	at org.micromanager.ndtiffstorage.NDTiffStorage$4.call(NDTiffStorage.java:929)
	at org.micromanager.ndtiffstorage.NDTiffStorage$4.call(NDTiffStorage.java:903)
	at org.micromanager.ndtiffstorage.NDTiffStorage$5.call(NDTiffStorage.java:1018)

henrypinkard, 2023-10-18:
I still cannot reproduce with those settings.
[image]

NDTiffReader line 307 catches an error from
```java
md = new JSONObject(getString(mdBuffer));
```

So maybe worth looking there to see if something is wrong with the metadata saved on disk, or if there is any metadata

It seems like the problem is its trying to zoom out, needs to further downsample the data to do so, but cannot read the existing data that it expects to be there. Maybe because of some missing synchronization between writing and reading? Hard to say

nicost, 2023-10-18:
That screenshot looks like it is from Explore mode.  Please select a grid (I use 3 x 3) and then press "Run Acquisition".  Maybe that was not clear with the initial phrase "running an acquisition"?

nicost, 2023-10-18:
Explore mode works for me as well, that is not the problem (other than that it has other problems).