micro-manager/micro-manager #1815: Micro-Magellan in explore mode kills the PFS on Nikon Ti2

nicost, 2023-10-13:
Presumably, the Explore mode sets the position of the Z Drive.  That action disables the Perfect Focus System on NIkon microscopes.  Explore mode works much better with the PFS active.  Can there be a setting that guarantees that the Z drive is not touched by the code?  Interestingly, this does not happen during the normal Magellan acquisitions.

henrypinkard, 2023-10-18:
Yes, that seems reasonable. Maybe you could add a checkbox or something here:
https://github.com/micro-manager/pycro-manager/blob/8f57c9f02a5e9fcc74582aabc076c83f050cf2b7/java/src/main/java/org/micromanager/explore/gui/ExploreControlsPanel.java#L143

And use it to exclude Z stuff from acquisition events here

https://github.com/micro-manager/pycro-manager/blob/8f57c9f02a5e9fcc74582aabc076c83f050cf2b7/java/src/main/java/org/micromanager/explore/ExploreAcquisition.java#L242

nicost, 2023-10-19:
Why a checkbox?  If the user does not ask for Z movement, why set the Z axis?

Also, why assign me?  It used to behave OK, some updates (that I can not oversee) broke its behavior, without any gain that I am aware of.  I would revert everything to the state it was in about 6 months ago.