micro-manager/micro-manager #1818: Support for Windows 11

ckolluru, 2023-10-13:
Do the nightly builds support Windows 11 OS? 

It seemed to work for me for a while, but now the application quits when I try to load a hardware configuration.

Thanks!

nicost, 2023-10-13:
Yes they do (at least, we have been using it on Windows 11 here).  Most likely, there is a problem in the hardware configuration. 

Try starting without a config file, go to` Help >  Create a Problem Report` start the report, then load your config file.  If that does not help you solve the issue, post the output on https://image.sc (tag with "Micro-Manager").

closed 2023-10-13 by nicost

chaitanya-kolluru-elucid, 2023-10-13:
Thanks for showing the logging mechanism, it was quite useful. 

Turns out it crashes when loading Spinnaker for which I had the wrong version installed (it needs 2.3.0.77 specifically as mentioned in the documentation, my bad!)

ckolluru, 2023-10-13:
Thanks for showing the logging mechanism, it was quite useful.

Turns out it crashes when loading Spinnaker for which I had the wrong version installed (it needs 2.3.0.77 specifically as mentioned in the documentation, my bad!)