micro-manager/micro-manager #1823: Snap-on-Move Preview doesn't remember movement devices

fadero, 2023-10-18:
After restarting micro-manager, the Snap-on-Move Preview doesn't remember which devices to poll. It'd be nice if I didn't have to re-add the devices each time I restart the software. 

Additionally, it'd be nice if I didn't have to remember to disable the Snap-on-Move Preview when I start an MDA. This may be too much to ask, so feel free to ignore this feature request if it's too difficult to implement.

marktsuchida, 2023-10-18:
Have you tried the newer "Snap after Move" checkbox in the Stage Control window? I'd be interested to know if you think we can remove the Snap-on-Move Preview plugin whose functionality is now (I think) redundant.

fadero, 2023-10-18:
I have tried the "Snap after Move" checkbox in the Stage Control window. It works fine.

That being said, I do _not_ think that the Snap-on-Move plugin is redundant because I have a physical joystick (for XY) and focus wheel (for Z) that I'd like to continue to use while Micro-Manager automatically snaps images while I steer around.

marktsuchida, 2023-10-18:
> I have a physical joystick (for XY) and focus wheel (for Z) that I'd like to continue to use while Micro-Manager automatically snaps images while I steer around.

Might the "polling update" checkbox in Stage Control enable this?

fadero, 2023-10-18:
It doesn't, at least with my current micro-manager build (2.0.3 20231013). It might be that I'm doing something wrong with the stage control, but it's never worked that way for me.

marktsuchida, 2023-10-18:
You're right! I just tested and it looks like the Polling Updates does work to update the positions displayed in the Stage Control window, and the Snap after Move does work to snap when a move is made by clicking a button in Stage Control, but the two do not work together: even with both checked, no snap is acquired if the stage is moved from outside of Stage Control (e.g. by a script command). @nicost Do you have any thoughts on whether this would be easy to fix (assuming the current behavior is not intentional)?

I'm happy to add memory to the Snap-on-Move Preview (with the caveat that it will probably need to discard such memory when the hardware config has changed), but we might end up in a better situation if we can make the simpler Stage Control handle all needed cases (it was already suggested previously by @nicost that we retire the Snap-on-Move Preview).

fadero, 2023-10-18:
Thanks for checking, Mark! If all of the other Snap-on-Move Preview features (e.g. polling interval, movement threshold) are also merged with the Stage Control functionality, then I'd support retiring Snap-on-Move Preview.

nicost, 2023-10-20:
The Stage Control plugin was not (yet) designed to have the same functionality a the Snap After Move plugin.  One important consideration (taken care of in the Snap After Move plugin) is the minimum amount of movement between two polling events that triggers a Snap.  We could move the UI and code over from the Snap After Move plugin, but that will make the Stage Control plugin even more complicated.  Thoughts how to do this in a simple way?

marktsuchida, 2023-10-20:
Hmm. Maybe it is a good thing to keep a separate UI. One way I can imagine is that we move the Snap-on-Move plugin into MMStudio (maybe accessible from the gear menu of the Preview image window) and have a "Snap on Move..." button (instead of the checkbox) in Stage Control. It might be good to simplify the Snap-on-Move UI if we do this, so that details are hidden by default.

Alternatively, we could turn the controller part of Snap-on-Move into something that can be activated from multiple UIs, so that when activated from Stage Control it automatically gets configured for the correct set of stages. The Stage Control window could then have the Snap on Move checkbox, together with a "Configure..." button to set the threshold.

In either case, it would probably be necessary to integrate the polling, too, because we don't want two things polling the same stages (probably won't matter for performance, but may be confusing and hard to debug). Seems a little complicated -- or at least there is real design work needed here.

Anyway, in the meantime, it turns out that Snap-on-Move already has some code that is supposed to save the settings, so I'll go ahead and see if I can fix that.

PR #1847 (2023-12-08): Snap on move improvements
Closes #1823.  Now remembers movement criteria between MM invocations.  Also automatically stops operation while an acquisition is running.

@marktsuchida, could you check if the way I handle pausing the monitoring thread is correct?  It works OK with the demo config.

I'd be happy to move this out of the beta menu item, probably into "Device Control".