micro-manager/micro-manager #1827: Multi-D Acquisition window freezes when changing window size

caichangjia, 2023-10-24:
Hi, I am using micromanager2.0 for Andor Zyla 5.5 camera on a Win 10 machine. I could run the following lines to capture a multi-D stack with full FOV size. However, when I switched to a smaller window size, the acquisition window will freeze and no output (see attached image).  
If I capture with Multi-D Acquisition GUI (instead of code) and the issue is not there.
Let me know if I missed anything, thank you!

```
core = Core()
TriggerMode = ['External', 'Internal (Recommended for fast acquisitions)'][1]
AcquisitionWindow = ['Full Image', '1024x1024', ' 512x512', ' 128x128'][3]
ElectronicShutteringMode = ['Global', 'Rolling'][0]
Overlap = ['On', 'Off'][0]
Exposure = 1.5

properties = {'TriggerMode': TriggerMode, 'AcquisitionWindow': AcquisitionWindow, 
              'ElectronicShutteringMode': ElectronicShutteringMode, 'Exposure': Exposure, 
              'Overlap': Overlap}#, 'FrameRate': FrameRate}
for key, value in properties.items():
    core.set_property('Andor sCMOS Camera', key, value)

save_name = r'acquisition_small_movie'
with Acquisition(directory=save_dir, name=save_name, ) as acq:
    events = multi_d_acquisition_events(num_time_points=400, time_interval_s=0)
    acq.acquire(events)
```
[image]

marktsuchida, 2023-10-24:
What exactly do you mean by "switched to a smaller window size"? Do you mean you resized the window with the mouse during (or after?) the acquisition?

@henrypinkard do you have any suggestions?

caichangjia, 2023-10-25:
Hi thanks for your reply! 
I mean first do the acquisition with AcquisitionWindow = 'Full Image', then do another round of capture with AcquisitionWindow = 128x128 using the following code (I do not resize window with mouse):
```
with Acquisition(directory=save_dir, name=save_name, ) as acq:
    events = multi_d_acquisition_events(num_time_points=400, time_interval_s=0)
    acq.acquire(events)
```
Hope that clarifies a bit.

marktsuchida, 2023-10-25:
Ah, I see. So the problem occurs when the cameras ROI is set using the `AcquisitionWindow` property.

It might be worth seeing if it make any difference if you set the ROI using `core.set_roi(x, y, width, height)` instead.

Also, does the problem only occur when an acquisition with a larger ROI was previously run? Does it make a difference if you first run with a small ROI and then switch to full sensor?

It would probably help to know your exact Micro-Manager (nightly) and pycro-manager versions, too (if they are not both the latest, please try the latest).

caichangjia, 2023-10-26:
Using core.set_roi still has the same problem. 
The problem happens every time when I change the ROI. But I could still do the multi-D Acq button in the GUI as normal. It just the command line no longer working.
My mm version is MMSetup_64bit_2.0.1_20230619, pycromanager version 0.27.5

marktsuchida, 2023-10-26:
@henrypinkard Would it be appropriate to move this issue to pycro-manager? Or NDViewer? Also any suggestions?