micro-manager/micro-manager #1849: Changing the format name of saved pictures with MM 2.0

Alex-code-lab, 2023-12-12:
Hello,
I'm still configuring my microscope on MM 2.0, but I have encountered a small problem: I can't manage to change the format name of my pictures. In the previous version, MM 1.4, the format was 'img_nbroftheframe_Channel.tif', and it might include additional information (like a z-axis, for example). In version 2.0, the format is quite different with more details, and I'm not sure how to modify it. Is there a solution?

Thank you for any input!

marktsuchida, 2023-12-12:
I'm afraid there is no way to modify the file name patterns currently.
The single-image file name patterns are:
- MM 1.4: `img_%09d_%s_%03d.tif`
    - `%09d`: time index
    - `%s`: channel name
    - `%03d`: Z index
- MM 2.x: `img_{}.tif`, where the `{}` is the axis indices (each prefixed by the axis name) separated by `_`
    - The axes are usually time, position, Z, and channel

Alex-code-lab, 2023-12-14:
Thank you very much for your answer. I will add a Python script to change those names.
Thank you again for your help!

closed 2023-12-14 by Alex-code-lab