micro-manager/micro-manager #1851: Routine to shut down lights at the end of an acquisition

Alex-code-lab, 2023-12-14:
Hello everyone,

When I run a multi-acquisition with my microscope, I would like all my lights to be turned off at the end of the routine. I have found the command lines in a script that allow me to turn them off:
"mmc.setConfig("Shutters Only", "All_off")
mmc.waitForConfig("Shutters Only", "All_off");
String currentState = mmc.getProperty("ZeissReflectedLightShutter", "State");
System.out.println("Current State: " + currentState);"

My goal is to have this executed at the end of the acquisition. I've heard about end of acquisition callbacks but haven't figured out how to do it. Maybe it's simpler to integrate it into an ".xml" script that I load each time? Maybe less comfortable but simpler? Let's note that I need it to still be really flexible as I need to change my gird of acquisition and my time step for each experiment.  
If anyone has an idea or a solution to help me out, that would be great.

Really sorry if I missed something here.
Thank you for your help, see you soon.

marktsuchida, 2023-12-14:
Usually, the best way is to ensure that the channel group used for MDA does _not_ contain the shutter states.
Instead, it should contain the `Core`-`Shutter` property, which specifies which shutter is active for each channel (preset).
Then, if Autoshutter is enabled, the correct shutter will open for each channel, and close after each.

Alex-code-lab, 2023-12-18:
Thank you for your help. However, the issue with the autoshutter is that it closes after each image capture. When I do a scan of different positions, if the shutter turns off between each position, it takes much more time. My goal was to have it turn off at the end of the image capture, before starting a new series, and then to close definitively at the end of the acquisition.

I have managed to make the shutter close at the end of the acquisition when I am doing only brightfield, which is already a good first step. But I am not sure if I can further complicate my acquisition to achieve my goals.

marktsuchida, 2023-12-18:
Indeed, we have a "keep shutter open" option for channels and Z stacks, but we don't have that for stage positions.

If you don't mind the user interface being a bit clunky, you could [run](https://micro-manager.org/apidoc/mmstudio/latest/org/micromanager/acquisition/AcquisitionManager.html#runAcquisition--) the MDA from your script:
```java
mm.acquisitions().runAcquisition();
print("Acquisition finished.\n");
```
This is equivalent to clicking the MDA "Acquire!" button, and will wait for the acquisition to finish. Then you can run your code to close the shutters.

closed 2025-04-21 by nicost