micro-manager/micro-manager #1856: Failed to load device "PE-800" from adapter module "CooLED"

zxliu117, 2023-12-25:
[image]

Hi, smart people! I am currently working on controlling CoolLED pE-800 using micromanager. It seems that the adapter module of pE-800 is too old compared to the nightly-build micro-manager 2.0. Is there any solution for this? I have tried the the stable or older release of micro-manager 2.0. They work but I am trying to write script using [pymmcore-plus](https://pymmcore-plus.github.io/pymmcore-plus/) or [pycro-manager](https://pycro-manager.readthedocs.io/en/latest/). It's really not convenient to use the old release of micro-manager.

Any help is precious to me. Thanks!

marktsuchida, 2024-01-06:
I assume you are getting `mmgr_dal_CoolLED.dll` from CoolLED?
pE300 and pE4000 are included in Micro-Manager but we do not have the source code for pE800.

You can either ask CoolLED to build their DLL for the current Micro-Manager interface, or (better yet) ask them to contribute their most current code to Micro-Manager (https://github.com/micro-manager/mmCoreAndDevices).

Cc: @CoolLEDMarketing.

closed 2024-01-06 by marktsuchida

CoolLEDMarketing, 2024-01-10:
Hello, I'm sorry you've had trouble. Please could you email support@coolled.com so we can assist you further.