micro-manager/micro-manager #1858: MDA slows significantly when displaying last image

fadero, 2024-01-03:
Tagging @nicost here because he also witnessed the issue. 

Essentially, when I am acquiring a timelapse with a large number of slices (>100 or so), I notice a gradual increase in the inter-timepoint delay _only when the display window is set to display the last image._ The inter-timepoint delay starts out normally (< 1 s), but it can increase to as much as 10 s when the display window is trying to display every image.

This significantly slows down the empty rate of the sequence buffer. The problem is ameliorated when I disable the display update. 

Is there any way to avoid this slowdown?

nicost, 2024-01-03:
Trying to reproduce this with the demo setup on my laptop with 4 cores, and no luck so far. Have been trying:
- 400 z slices, set "UseSequences" to Yes
- DemoStateDevice only for 3 channels, set "Sequence" to "On", exposure times to 2 ms
- Set DemoCamera-FastImage to "1".
- Use different image sizes (512x512, 1980x680)
- Save images to local NVME SSD drive
So far, I can only get to situations where the display slows down over time (gets choppy), which can somewhat be ameliorated by switching off histogram update (which you may already had switched off). What was yor image size?  Any other obvious things I may have forgotten about?

fadero, 2024-01-03:
Image size is usually ~1960x256. I had ~1000 z slices and 3 channels with 2 ms exposures. 

Do you have it running such that MDA acquires all channels at every slice?

nicost, 2024-01-03:
I have done it both ways.  However, for whatever reason, my democamera does not go faster than 85fps in live mode (independent of ROI that I set).  I am not sure whether that is a software issue with the democamera adapter or a hardware issue with my laptop.  Will look at the democamera code and test tomorrow with workstation upstairs.

marktsuchida, 2024-01-06:
I would note that very subtle differences in the camera frame rate and popping rate can swing the sequence buffer between the two extremes of almost-always-empty and overflow. I tried with demo camera 1960x256 FastImage, 1000 Z slices (UseSequences), no timepoints or channels, save stack file to NVMe:

| Exposure (ms) | Sequence Buffer Monitor |
| --- | --- |
| 1.6 | overflows after popping around 800-850 frames (261-frame capacity) |
| 1.7 | gradually hovers up to 40 frames or so occupied |
| 1.75 | never exceeds 10 frames occupied |

With the 1.7 ms exposure, the sequence buffer occupancy did seem to go up gradually. Also, when 10 timepoints (0 interval) were added, the buffer filled up more, even at 1.75 ms exposure. This was also somewhat the case with 10,000 timepoints and no Z slices. In general, popping from the buffer seems to slow down as the size of the dataset increases and especially if there are multiple axes (but this was a much more subtle effect than adding 10 s between time points).

Adding 3 channels (only switching demo LED, UseSequences), in my case, caused the display to almost stop updating (even with histograms Never; this could be a bug), but I could not find conditions that would visibly slow down the draining of the sequence buffer (how many time points did you have?). For this experiment (10 time points, 1000 slices, 3 channels, 1.7 ms exposure) I used a larger sequence buffer (1044 frames) and it filled up to ~350 frames max. The MDA (I had it set to the Clojure one) drained the buffer after each time point (3,000 planes) but the max buffer occupancy seemed to plateau for me. My 6-core (12 virtual core) PC did not reach 100% CPU (it was about 75-80% in Task Manager > Performance). If I ran a Micro-Manager C++ build (reaches 100% CPU on its own) during the acquisition, the sequence buffer quickly overflowed (unsurprsingly).

The time it takes to drain the sequence buffer after each time point reflects how full the buffer got, which is probably quite sensitive to small changes in popping rate. Also, the display is not supposed to directly limit the rate at which the buffer is drained, only indirectly by consuming CPU.

I'd expect that for any given computer there will be some maximum frame rate (given all other settings) above which everything gets bogged down. It's probably already starting to push it if the sequence buffer is operating in anything but the almost-always-empty regime, although it is understandable that this is necessary for fast burst sequences.

There may, however, be room for improvement if the drastic slowdown happens despite the CPU being far from saturated.

fadero, 2024-01-06:
>  but I could not find conditions that would visibly slow down the draining of the sequence buffer (how many time points did you have?)

We notice an inter-timepoint slowdown after fewer than 10 timepoints. @nicost replicated the issue on our system yesterday and he seems to have identified the root cause, but I'll let him explain his thinking (because I don't fully understand it).

nicost, 2024-01-08:
There are many "subtle" issue here.  I'll try to split them out.
1. It surprises me that the sequence buffer is drained after each time point.  We saw this with both the Clojure and the Java Acquisition Engines.  Is this a feature of the engines?  This is not necessarily bad, but it surprised me, and me be a problem for experiments where timing is critical.
2. I could never get the Democamera to deliver images at high enough rates.  The code uses a CDeviceUtils::Sleepms(1) function and then checks if the exposure time has passed.  I believe that - at least on my Windows machine - that Sleep function results in ~12 ms delay before the Thread is resumed.  This may well be different on a Mac, but we may want to have a better mechanism to emulate very fast acquisitions.
3. The display slows down with more images, concomitant with accumulation of images in the circular buffer. I doubt this is due to lack of resources, as Tanner's computer has something like 24 cores (which were kept almost equally busy, but the total CPU activity did not go over ~ 30%), but even if it is, it is worthwhile addressing this slow down in viewer updating.  This is much more obvious in multi-channel acquisitions than in single channel acquisitions.  I believe the root cause is in the functions :  Storage::getImagesIgnoringAxes(Coords coords, String... ignoreTheseAxes) and Storage::getImagesMatching(Coords coords).  The RAMM and MultiPageTiff implementations for both functions loop through all known Coords to see if they match.  The function getImageIgnoringAxes optimizes single axes acquisitions (it can simply return the Coord in the function call), explaining why that situation is so much faster.   I started working on optimizing these (by using a hashmap for Coords with the Coord stripped from its channel as the key) in this branch: https://github.com/nicost/micro-manager/tree/storageIndexCoords.  One issue that confuses me is that these function are very similar, yet subtly different.  I may see what is involved in deprecating "getImagesMatching". 

Thoughts appreciated!

marktsuchida, 2024-01-08:
> * It surprises me that the sequence buffer is drained after each time point.  We saw this with both the Clojure and the Java Acquisition Engines.  Is this a feature of the engines?  This is not necessarily bad, but it surprised me, and me be a problem for experiments where timing is critical.

I don't know if this is intentional or accidental in the design, but it does prevent the issue of taking a lot of time to get an acquisition started due to having to generate and merge all the frames.

> * I could never get the Democamera to deliver images at high enough rates.  The code uses a CDeviceUtils::Sleepms(1) function and then checks if the exposure time has passed.  I believe that - at [...]

I was testing on Windows. I don't get above 100 fps with DemoCamera FastImage (512x512, 2 ms exposure) -- in fact it seems to be reproducibly 99-101 fps. You might try SequenceTester (although this is not its intended use case). With it I get ~7000 fps (in Live mode). (This definitely explains why I could not reproduce the effects you were seeing with the real camera.)

> * ... I believe the root cause is in the functions :  Storage::getImagesIgnoringAxes(Coords coords, String... ignoreTheseAxes) and Storage::getImagesMatching(Coords coords).

Potentially related: https://github.com/micro-manager/micro-manager/pull/1250

I do not know the reason for introducing `getImagesIgnoringAxes` but that was in f792a03e20742e1b2098614087078ca1e177d58a.

nicost, 2024-01-08:
> I don't know if this is intentional or accidental in the design, but it does prevent the issue of taking a lot of time to get an acquisition started due to having to generate and merge all the frames.

Not sure I understand.  What frames need to be generated and merged?  What part of the code is doing this?  Do you have any idea where in the code this wait for the sequence buffer to empty after each time point is taking place?

> Potentially related: https://github.com/micro-manager/micro-manager/pull/1250

Yes, completely the same issue.  I will soon generate a PR from branch https://github.com/nicost/micro-manager/tree/storageIndexCoords after some testing.  This branch nicely speeds up the display for large datasets (by using a Hash code lookup for Coors excluding the channel axis). 

> I do not know the reason for introducing getImagesIgnoringAxes but that was in https://github.com/micro-manager/micro-manager/commit/f792a03e20742e1b2098614087078ca1e177d58a.

Yes that was me;).  Pretty sure I did this because I failed to understand the documentation to `getImagesMatching()`.  The only place I see `getImagesMatching` used is in the Album code.  Readability of the code is much better when being explicit about the axes to be ignored (which also may have become more of an issue with equating the absence of an axis with that axis having value zero).

marktsuchida, 2024-01-08:
> Not sure I understand. What frames need to be generated and merged? What part of the code is doing this? Do you have any idea where in the code this wait for the sequence buffer to empty after each [...]

I'm talking about the internals of (both) acquisition engines, and the problem that came up (most recently) in https://forum.image.sc/t/several-minute-delay-before-acquisition-starts/90068. This is probably not a reason to justify breaking up the acquisition by timepoint, but it is something that needs to be considered if changing the engines not to pause between timepoints, because the sequence acquisition lengths may increase by a couple orders of magnitude for many users.

I don't have a strong opinion on `getImagesMatching`; perhaps it can be implemented in terms of `getImagesIgnoringAxes`, which would be great; otherwise it can be deprecated or a note about poor performance could be added. I agree that `getImagesIgnoringAxes` seems like a better API for typical use cases.

nicost, 2024-01-08:
> I'm talking about the internals of (both) acquisition engines, and the problem that came up (most recently) in https://forum.image.sc/t/several-minute-delay-before-acquisition-starts/90068. This is [...]

Ah yes!  That still does not (fully) explain in my mind why the acquisition engines wait starting the next time point until the first one is completely purged from the sequence buffer.  At least in theory, these two things could be uncoupled.  Probably not very important right now, but I may look at this in the future.

marktsuchida, 2024-01-09:
> why the acquisition engines wait starting the next time point until the first one is completely purged from the sequence buffer

It has to, because the moment you start a new sequence acquisition, the Core will clear the sequence buffer (we can't change this in our current MMCore API because existing code would then see unintentionally leftover images). The only alternative is to run all as a single sequence acquisition.

PR #1862 (2024-01-10): Optimizes Storage:getImagesIgnoringAxes implementations.
Closes #1858. The implementations of the function Storage:getImagesIgnoringAxes would iterate over all known coords to check if they match or not.  This is an expensive operation, especially for large datasets.  Our code only uses this function with the axis Channels as the ignored axes, so I added code to store those in a HashMap for fast retrieval.  This change avoids the slow down in display speed for large datasets.

The function GetImagesMatching appears to be almost the same in intent.  I changed its implementations to call getImag [... cut]
closed 2024-01-10 by nicost