micro-manager/micro-manager #1868: Camera seems to crash when pycromanager calls core.snap_image() more than once

mdanderson03, 2024-01-11:
I have used pycromanager for years and never had this issue. I was running it earlier today and it was working well and then nothing changed and it wouldnt work anymore. The issue I am having is when micromanager is first loaded up, I can use MM core to snap and image one time. If I ask it again in the same script or rerun the script, it crashes. I have MM and pycromanager up to date. I have an IRIS 15 camera with up to date driver. I have restarted software and hardware and it still does it. Its almost like it forgets about the camera after I initially ask it to snap an image. The core log says: 'The acquisition setup is needed first (PL_ERR_ACQUISITION_SETUP_REQUIRED)'. I dont know how to set that up though. 

[image]

[image]

[corelog_snap_issue.txt](https://github.com/micro-manager/micro-manager/files/13908008/corelog_snap_issue.txt)

Code for reproduction

# your code here
from pycromanager import Core

core = Core()
core.snap_image()

**Expected outcome**
Take a snap shot

**Actual outcome**
First try takes a snap shot. Will never take another one if rerun

mdanderson03, 2024-01-11:
Also the core log says Error in device "FilterWheel:0:38": Invalid state (position) requested (12). I literally never ask it to go to state 12. Its impossible to as it only goes to 8.

marktsuchida, 2024-01-11:
Let's try to isolate the issue by eliminating pycro-manager and ASITiger, since the error message seems to come from PVCAM (but errors from ASITiger are logged).
Can you create a test hardware configuration that contains only the camera, and see if it has the same error? And do you get the same error when snapping by clicking the Snap button (as opposed to via `snap_image()`)?

mdanderson03, 2024-01-12:
Mark,

Appreciate the reply. I've been studying this more and I came across a
thread in image.sc about snap_image. I was unaware that asking to
snap_image again without retrieving the image will cause issues. I went to
this low level function in hopes to isolate the problem better. I updated
and restarted everything, but tested it with only doing the snap_image
function which I now believe was destined to fail as a test. Tomorrow I'm
going to try it again with retrieving the image as well and see if it does
it. If it fails, I will do your suggestion and have just camera in the
build and report back with the core log and results.

mdanderson03, 2024-01-12:
Also, I can hit the snap button in the gui as many times as I want and it
never does it. After the command is sent from pycromanager and gives back
that error message back, the snap button will stop working though.

mdanderson03, 2024-01-12:
Update: it was the issue of not having the get_tagged_image() run after snap_image().

closed 2024-01-12 by mdanderson03